boatbod/op25 #153: frequency is not the same on web interface and cannot be changed

awipslinuxuser, 2022-05-15:
so i have my local fire station  154.41500  programmed into op25 but when i launch the web interface its sho
[image]
[image]
wing this frequency 0.000154 and i cannot change the frequency but it should be the same as from the terminal is

awipslinuxuser, 2022-05-15:
and yes i have tried to use skip and the arrows but the frequency didn't change

thetoivonen, 2022-05-16:
Input frequency in Hz not MHz so change to 154.415e6

closed 2022-06-09 by boatbod