boatbod/op25 #155: What does `process_data_unit timeout` mean?

devicenull, 2022-06-05:
I regularly see a bunch of `process_data_unit timeout` entries in the log - is this indicating a problem at all?  I can't quite wrap my head around what the code is attempting to do here.

boatbod, 2022-06-09:
That message only comes from trunking.py, so I assume you are running the rx.py version of op25.
In a nutshell it's just an informational message that a timeout has occurred either on a voice channel or a control channel.  You'd need to post some of the actual log for me to be more specific about what's going on, but in general I wouldn't worry too much about it... there are lots of messages that are output at log level 10 & 11 simply for debugging purposes when it becomes necessary.

closed 2022-06-09 by boatbod

devicenull, 2022-06-09:
I'm running this through trunk-recorder, so I wasn't sure how relevant this was to anything else.  Regardless, for anyone else that finds this I stopped seeing this message once I moved my antenna outside and (presumably) started getting way better signal strength.