boatbod/op25 #163: After running great for a year, I get the following error:

Charlier26, 2022-09-28:
I had it running great for a year, and suddenly when I start op25, I get:

osmosdr source_c creation failure
Traceback (most recent call last):
  File "./rx.py", line 934, in <module>
    rx = rx_main()
  File "./rx.py", line 848, in __init__
    self.tb = p25_rx_block(self.options)
  File "./rx.py", line 125, in __init__
    gain_names = self.src.get_gain_names()
AttributeError: 'NoneType' object has no attribute 'get_gain_names'

I tried to reinstall, using the install.sh file, but keep getting the same error.

Am I missing something?

Charlier26, 2022-09-28:
I rebuilt the entire pi. I now have it playing great into the local speaker (on the jack), but am having problems setting up either darkice or liquidsoap to stream it to my icecast2 server. Apparently there is some encoder that doesn't install with liquidsoap, and I'm trying to find a configuration file to use with darkice and op25.

boatbod, 2022-10-17:
Over time the liquidsoap audio functions have changed in small ways.  The example op25.liq file should be good for any version.

closed 2023-01-13 by boatbod