boatbod/op25 #166: decrypt dmr

romanremus, 2023-01-06:
hello Boatbod ..... in my country I can only enjoy a few dmr signals ... which are clear and have a very rare activity, but the other dmr signals that have a permanent activity are with AD RC4, being a radio amateur I'm listening.....please, do you think you can make it work on windows too??? or can it be added to dsd somehow. or how can I use your software??? thank you very much

boatbod, 2023-01-09:
You could consider running ubuntu in a virtual machine under windows, but there's no way for me to do a native port of op25 into a windows environment.

closed 2023-01-09 by boatbod

romanremus, 2023-01-09:
How can comand for start dmr witch key RC4 ?.....thanks 
    Pe luni,