boatbod/op25 #168: Too many errors compiling for gr310

srs4511351, 2023-01-18:
I have a Raspberry Pi 4 running 64 bit Debian Bookworm.

When compiling op25,gr310 branch, there are many fatal and other errors and it finally fails.

cd ~
git clone https://github.com/boatbod/op25
cd op25
git checkout gr310
./install.sh

These are samples of the many errors:

/usr/include/c++/12/new:166:6: note: non-usual
      'operator delete' declared here
void operator delete(void*, std::size_t, std::align_val_t)
~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.

CMake Error at /usr/lib/aarch64-linux-gnu/cmake/gnuradio/GrPybind.cmake:214 (list):
  list GET given empty list
Call Stack (most recent call first):
  op25/gr-op25_repeater/python/op25_repeater/bindings/CMakeLists.txt:47 (GR_PYBIND_MAKE_OOT)

-- Configuring incomplete, errors occurred!
See also "/home/pi/op25/build/CMakeFiles/CMakeOutput.log".
make: *** No targets specified and no makefile found.  Stop.
make: *** No rule to make target 'install'.  Stop.

I can attach the whole long list if necessary.

boatbod, 2023-01-22:
Those errors are caused by gcc-12 incompatibility with gnuradio's gr_modtool.
https://github.com/gnuradio/gnuradio/issues/6477

If you can manually install the package for gcc-11 and use that instead of gcc-12 it should build ok.

closed 2023-01-22 by boatbod

srs4511351, 2023-01-23:
It looks like you opened an issue on gnuradio about this.
I hope a graceful solution can be found.
On Debian bookworm, I am finding it impossible to get gcc-12 removed (dependency issues) and gcc-11 installed.
With Debian bookworm, gcc-12 comes with gnuradio 3.10.
My only alternative short of getting help doing this, is to try a Debian Bullseye system and install gnuradio 3.10 on that.

srs4511351, 2023-01-24:
In the Bullseye system,
I changed the executable link for gcc to point to gcc-11 and many of the errors went away, but it still failed.

Booting on another SD card,
I installed gnuradio 3.10.5.0 from source on a Debian Bullseye system.
It has gcc 10.2.1, c++ 10.2.1, g++ 10.2.1
install.sh detected the package version of gnuradio 3.8 that was not installed, so I installed it with the commands that I use for the gnuradio 3.8 version.

cd ~
git clone https://github.com/boatbod/op25
cd op25
git checkout gr310
sudo sed -i -- 's/^# *deb-src/deb-src/' /etc/apt/sources.list
echo "/usr/bin/python3" > op25/gr-op25_repeater/apps/op25_python

mkdir build
cd build
cmake ../

These are the last few errors. They look exactly like the patched Bookworm results with gcc-11.

CMake Error at /usr/local/lib/cmake/gnuradio/GrPybind.cmake:213 (list):
  list GET given empty list
Call Stack (most recent call first):
  op25/gr-op25_repeater/python/op25_repeater/bindings/CMakeLists.txt:47 (GR_PYBIND_MAKE_OOT)

CMake Error at /usr/local/lib/cmake/gnuradio/GrPybind.cmake:214 (list):
  list GET given empty list
Call Stack (most recent call first):
  op25/gr-op25_repeater/python/op25_repeater/bindings/CMakeLists.txt:47 (GR_PYBIND_MAKE_OOT)

-- Configuring incomplete, errors occurred!
See also "/home/pi/op25/build/CMakeFiles/CMakeOutput.log".
See also "/home/pi/op25/build/CMakeFiles/CMakeError.log".

boatbod, 2023-01-24:
Generally speaking you cannot safely use the install.sh script with a 
built-from-sources gnuradio because the install script will detect only 
the package version of gnuradio not the locally built one. There is the 
issue of mismatched libraries such as package installed gr-osmosdr and 
librtlsdr which will fail because they were built against gr-3.8 unless 
you locally build and install copies of them too.

Standard Bullseye works fine with gnuradio-3.8 and the op25 install.sh 
script.

On 1/24/23 02:23, srs4511351 wrote:
> In the Bullseye system,
> I changed the executable link for gcc to point to gcc-11 and many of
> the errors went away, but it still failed.
>
> Booting on another SD card,
> I installed gnuradio 3.10.5.0 [...]

boatbod, 2023-01-24:
> I built gr-osmosdr against gnuradio 3.10 from source.
> The built from source gnuradio 3.10 had the same or similar failures as the one installed from packages.
> Do you see anything in particular that caused my failures?
> I could build build librtlsdr from source if that would help.
> We need to be able to build op25 with gnuradio 3.10, as the end of 3.8 is drawing near.

To give yourself the best chance of success, apt purge everything gnuradio related then restart the build from cmake on up.  Don't use install.sh

srs4511351, 2023-01-24:
I purged every package that I could find related to gnuradio. Still the same result, but I looked deeper into the errors. One concern is that decoder_bf_python.cc is missing.
On github, the comments by most files in op25/op25/gr-op25/python/op25/bindings/ refer to restoring deleted files...
...
-- Found pybind11: /usr/include (found version "2.6.2" )
Traceback (most recent call last):
  File "/home/pi/op25/op25/gr-op25/python/op25/bindings/header_utils.py", line 69, in <module>
    pbhp = PybindHeaderParser(args.pathname)
  File "/home/pi/op25/op25/gr-op25/python/op25/bindings/header_utils.py", line 9, in __init__
    with open(pathname, 'r') as f:
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/op25/op25/gr-op25/python/op25/bindings/decoder_bf_python.cc'
CMake Error at /usr/local/lib/cmake/gnuradio/GrPybind.cmake:211 (list):
  list GET given empty list
Call Stack (most recent call first):
  op25/gr-op25/python/op25/bindings/CMakeLists.txt:39 (GR_PYBIND_MAKE_OOT)

boatbod, 2023-01-24:
The bindings files normally get generated by the install.sh script but 
you can manually achieve the same thing by running ~/op25/build_bindings.sh

srs4511351, 2023-01-25:
Do we need the missing file decoder_bf_python.cc ?

srs4511351, 2023-01-25:
I tried the modified procedure in README-gr3.10, but it still seems that gr_modtool bind is still failing in build_bindings.sh.
I read that it compiles perfectly in some OSes, so there must be something wrong with mine, other dependencies and versions. I guess I'll have to wait for things to be straitened out.

Building bindings for gr-op25
GNU Radio module name identified: op25
creating docstrings directory /home/pi/op25/op25/gr-op25/python/op25/bindings/docstrings
ERROR error occured, while parsing element with name "Field" and attrs "['id', 'name', 'type', 'context', 'access', 'offset']".
Error: 'line'.
'line'

Thanks.

boatbod, 2023-01-25:
What was your base os?
Debian Buster should be good, as is Ubuntu 20.04, 22.04 and Raspberry PI 
OS (current ver).

srs4511351, 2023-01-26:
I tried two operating systems.

2023.01.01 **bookworm** from Debian. Installed gcc-11. update-alternatives did not offer any choices, so I switched the gcc link to point to gcc-11, g++ link to point to g++-11, cpp link to point to cpp-11. I hope I didn't miss one.

2023.01.02 **bullseye** from Debian Installed gnuradio 3.10.5.0 from source.

They both fail on gr_modtool bind

I could try a raspios-bullseye system with gnuradio 3.10.5.0 installed from source if you think it will help.
Buster is quite old.

boatbod, 2023-01-26:
Why can't you just run Bullseye without building gnuradio from sources?  
As I recall it's a gnuradio-3.8 release so it'll work just fine.
Bookworm is definitely no-go right now.

srs4511351, 2023-01-26:
op25 does work with GR 3.8 and Bullseye.
The point of this exercise is to get everything working on Bookworm/GR 3.10.
I have many SDR applications installed and Bookworm with GR 3.10 will be released.
It is time for me to sit back and wait for this to be fixed, although for some reason, it does work for some OSes.

srs4511351, 2023-02-12:
I see that you provided pre-made bindings files.
I also noted the issue with the compiler versions.
I was able to compile and install op25 on my Raspberry Pi bookworm system with gr 3.10.5.1
I just omitted the build_bindings.sh step.

There were problems, though.
Just before I installed op25, apt did a bunch of upgrades, including gnuradio.
When I initially ran it, I got a  'osmosdr source_c creation failure' message.
I recompiled and installed gr-osmosdr. After that, op25 ran and functioned!

I don't know if it is a valid comparison, but gqrx, which was compiled before the gnuradio upgrade, ran just fine after the upgrade and also after the gr-osmosdr reinstall.

I hope that reinstalling things every time there are upgrades does not become the standard.