boatbod/op25 #169: GOTO Not working while tracking multiple sites

Woody929, 2023-01-20:
If I only have OP25 set to monitor a single site of a system, the GoTo functionality works.   However if the cfg.json has multiple sites enabled the function no longer holds on the group id specified.

boatbod, 2023-01-22:
In multi_rx.py, commands such as lockout and goto are issued to a the system being tracked on the terminal.  Before I ask for level 11 debug logs, please confirm you have the terminal displaying the particular system you are trying to invoke a "goto" on?  (In the curses terminal the left/right arrow keys cycle between system.  On the web terminal the small "<" ">" buttons flanking the "d" on the right of the page do the same job.  
Assuming you've checked that, please turn the log level up and capture the scenario so I can take a look at what's going on.
Thanks, Graham

Woody929, 2023-01-25:
After rebooting everything, I can no longer duplicate this issue.  In the event I see it again, I'll boost the logging level and open another ticket.  Thanks for all you do.

closed 2023-01-25 by Woody929