boatbod/op25 #170: Audio dropout

gcurtis79, 2023-01-25:
I've been using your fork for a couple years now, and recently started having issues with "audio closing". I have 2 raspberry pi units streaming to Broadcastify with darkice, and only one of them seems to have this issue.

Sometimes it will work properly for hours, sometimes only minutes, but this is the log from rx.py. if I need to get more info, let me know.

```
streaming server url="http://audio3.broadcastify.com:80/########.m3u"
using ALSA sound system
audio device: loop1
Listening on 127.0.0.1:23466
python version detected: 3.9.2 (default, Feb 28 2021, 17:03:44) [GCC 10.2.1 20210110]
Allocating 15 zero-copy buffers
audio closing
```

boatbod, 2023-01-25:
Seeing an "audio closing" message out of the blue means that the 
sockaudio thread encountered an error and decided to quit.  There's a 
number of places the error could occur, and unfortunately right now none 
produce any logging to give us a hint.  As soon as I get an opportunity 
I'll add some additional debug and we can do more troubleshooting.

boatbod, 2023-01-26:
I just noticed that you are streaming via Darkice, which I really don't recommend especially on an RPi.  Suggest you switch over to liquidsoap and see how that goes.  There's some basic instructions in the ~/op25/README-rpi3-liquidsoap file

gcurtis79, 2023-01-26:
I've tried liquid soap before, and ran into issues. I tried again just now, and I'm getting an error.

I created sara.liq only changing my icecast parameters for broadcastify settings, and after running sara.liq I get:

`
[dynamic.loader:3] Could not find dynamic module for fdkaac encoder.
`

And sara.liq exits.

I can't find a package to satisfy this.

gcurtis79, 2023-01-26:
Looks like I think I got it. While testing, I uninstalled liquidsoap through apt, and installed it with opam. This fixed the fdkaac issue, but it still closed. I commented out the line:
`
out (input)
`
And it started working.

Thank you.

boatbod, 2023-01-26:
You probably do have to turn off local default audio output, especially 
if you're on a RPI and the default is trying to use Pulse Audio which 
typically isn't running during a headless boot.

gcurtis79, 2023-01-26:
I might test doing just that on my other setup if it starts having issues, but the fix I ended up with was this from another project that has similar errors:

`
sudo apt install opam screen
opam init
opam switch create 4.10.0
opam depext taglib mad lame ffmpeg vorbis cry samplerate ocurl liquidsoap fdkaac alsa
opam install taglib mad lame ffmpeg vorbis cry samplerate ocurl liquidsoap fdkaac alsa
`

I'm not sure why it was trying to use aac in the first place, unless the stream from op25 originates as aac. But this and turning off local audio solved my issue.

I appreciate the quick responses. You can consider this issue closed.

closed 2023-01-26 by gcurtis79