boatbod/op25 #172: ValueError: Unable to coerce endpoints: 'gnuradio.op25_repeater.op25_repeater_python.gardner_cc' object is not subscriptable

mldulaney, 2023-01-29:
I am on the gr310 branch.  I am getting the following failure:

Traceback (most recent call last):
  File "/home/s/rpmbuild/BUILD/op25-gr310/op25/gr-op25_repeater/apps/./multi_rx.py", line 1085, in <module>
    rx = rx_main()
  File "/home/asdf/rpmbuild/BUILD/op25-gr310/op25/gr-op25_repeater/apps/./multi_rx.py", line 1048, in __init__
    self.tb = rx_block(options.verbosity, config = byteify(config))
  File "/home/asdf/rpmbuild/BUILD/op25-gr310/op25/gr-op25_repeater/apps/./multi_rx.py", line 594, in __init__
    self.configure_channels(config['channels'])
  File "/home/asdf/rpmbuild/BUILD/op25-gr310/op25/gr-op25_repeater/apps/./multi_rx.py", line 766, in configure_channels
    chan = channel(cfg, dev, self.verbosity, msgq_id, self.rx_q, self)
  File "/home/asdf/rpmbuild/BUILD/op25-gr310/op25/gr-op25_repeater/apps/./multi_rx.py", line 216, in __init__
    self.demod = p25_demodulator.p25_demod_cb(
  File "/home/asdf/rpmbuild/BUILD/op25-gr310/op25/gr-op25_repeater/apps/p25_demodulator.py", line 430, in __init__
    self.connect_chain(demod_type)
  File "/home/asdf/rpmbuild/BUILD/op25-gr310/op25/gr-op25_repeater/apps/p25_demodulator.py", line 504, in connect_chain
    self.connect(self.if_out, self.cutoff, self.agc, self.fll, self.clock, self.diffdec, self.costas, self.to_float, self.rescale, self.slicer)
  File "/usr/lib64/python3.10/site-packages/gnuradio/gr/hier_block2.py", line 36, in wrapped
    raise ValueError("Unable to coerce endpoints: " + str(err))
ValueError: Unable to coerce endpoints: 'gnuradio.op25_repeater.op25_repeater_python.gardner_cc' object is not subscriptable

cwd is ~/rpmbuild/BUILD/op25-gr310/op25/gr-op25_repeater/apps

Trying to run ./multi_rx.py -v 1 -c psern.json

boatbod, 2023-01-29:
Not seen that error before.  Please let me know more about your o/s version, gnuradio version, gcc version and how you built op25.
i.e. Ubuntu 22.04, installed gnuradio-3.10 from packages, built using gcc-11,etc.

mldulaney, 2023-02-05:
Hi, sorry for the delay in responding.  Life, and all.

I'm on Fedora 36, gnuradio-3.10.1.0-2.fc36.x86_64, gcc version 12.2.1.  I'm trying to build it as an rpm.

The relevant details:

Executing(%build): /bin/sh -e /var/tmp/rpm-tmp.4h7J4U
+ umask 022
+ cd /home/mdulaney/rpmbuild/BUILD
+ CFLAGS='-O2 -flto=auto -ffat-lto-objects -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1  -m64  -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection'
+ export CFLAGS
+ CXXFLAGS='-O2 -flto=auto -ffat-lto-objects -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1  -m64  -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection'
+ export CXXFLAGS
+ FFLAGS='-O2 -flto=auto -ffat-lto-objects -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1  -m64  -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -I/usr/lib64/gfortran/modules'
+ export FFLAGS
+ FCFLAGS='-O2 -flto=auto -ffat-lto-objects -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1  -m64  -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -I/usr/lib64/gfortran/modules'
+ export FCFLAGS
+ LDFLAGS='-Wl,-z,relro -Wl,--as-needed  -Wl,-z,now -specs=/usr/lib/rpm/redhat/redhat-hardened-ld -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1  -Wl,--build-id=sha1 -Wl,-dT,/home/mdulaney/rpmbuild/BUILD/op25-gr310/.package_note-op25-0.1-1.fc36.x86_64.ld'
+ export LDFLAGS
+ LT_SYS_LIBRARY_PATH=/usr/lib64:
+ export LT_SYS_LIBRARY_PATH
+ CC=gcc
+ export CC
+ CXX=g++
+ export CXX
+ '[' -f /usr/lib/rpm/generate-rpm-note.sh ']'
+ /usr/lib/rpm/generate-rpm-note.sh op25 0.1-1.fc36 x86_64
+ cd op25-gr310
+ ./build_bindings.sh
Building bindings for gr-op25

<snip build_bindings.sh output>

Writing binding code to python/op25_repeater/bindings/docstrings/ysf_tx_sb_pydoc_template.h
+ CFLAGS='-O2 -flto=auto -ffat-lto-objects -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1  -m64  -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection'
+ export CFLAGS
+ CXXFLAGS='-O2 -flto=auto -ffat-lto-objects -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1  -m64  -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection'
+ export CXXFLAGS
+ FFLAGS='-O2 -flto=auto -ffat-lto-objects -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1  -m64  -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -I/usr/lib64/gfortran/modules'
+ export FFLAGS
+ FCFLAGS='-O2 -flto=auto -ffat-lto-objects -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1  -m64  -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -I/usr/lib64/gfortran/modules'
+ export FCFLAGS
+ LDFLAGS='-Wl,-z,relro -Wl,--as-needed  -Wl,-z,now -specs=/usr/lib/rpm/redhat/redhat-hardened-ld -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1  -Wl,--build-id=sha1 -Wl,-dT,/home/mdulaney/rpmbuild/BUILD/op25-gr310/.package_note-op25-0.1-1.fc36.x86_64.ld'
+ export LDFLAGS
+ LT_SYS_LIBRARY_PATH=/usr/lib64:
+ export LT_SYS_LIBRARY_PATH
+ CC=gcc
+ export CC
+ CXX=g++
+ export CXX
+ /usr/bin/cmake -S . -B redhat-linux-build -DCMAKE_C_FLAGS_RELEASE:STRING=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE:STRING=-DNDEBUG -DCMAKE_Fortran_FLAGS_RELEASE:STRING=-DNDEBUG -DCMAKE_VERBOSE_MAKEFILE:BOOL=ON -DCMAKE_INSTALL_DO_STRIP:BOOL=OFF -DCMAKE_INSTALL_PREFIX:PATH=/usr -DINCLUDE_INSTALL_DIR:PATH=/usr/include -DLIB_INSTALL_DIR:PATH=/usr/lib64 -DSYSCONF_INSTALL_DIR:PATH=/etc -DSHARE_INSTALL_PREFIX:PATH=/usr/share -DLIB_SUFFIX=64 -DBUILD_SHARED_LIBS:BOOL=ON

boatbod, 2023-02-06:
Might be related to your version of pybind11?
I did find this, but not sure if totally relevant since the existing code works on other distros.
https://github.com/gnuradio/gnuradio/issues/4773

boatbod, 2023-02-07:
Two things I've learned over the last 24hrs... both the pybind11 version and compiler version need to be the same for both gnuradio and op25.  You can check what was used for gnuradio as follows:
[code]
gnorbury@vaio:~/op25/op25/gr-op25_repeater/apps$ gnuradio-config-info --pybind
2.9.1
gnorbury@vaio:~/op25/op25/gr-op25_repeater/apps$ gnuradio-config-info --cc
cc (Ubuntu 12.1.0-2ubuntu1~22.04) 12.1.0 
Copyright (C) 2022 Free Software Foundation, Inc. 
This is free software see the source for copying conditions.  There is NO 
warranty not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
[/code]

mldulaney, 2023-02-10:
```
♥️ gnuradio-config-info --pybind
2.9.2
♥️ gnuradio-config-info --cc
gcc (GCC) 12.1.1 20220507 (Red Hat 12.1.1-1) 
Copyright (C) 2022 Free Software Foundation, Inc. 
This is free software see the source for copying conditions.  There is NO 
warranty not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
♥️ rpm -q pybind11-devel
pybind11-devel-2.9.2-1.fc36.x86_64
```

mldulaney, 2023-02-10:
ahhh, gcc is newer

mldulaney, 2023-02-11:
Welp, still getting the same error despite rebuilding gnuradio against the same gcc
gnuradio-config-info --cc
gcc (GCC) 12.2.1 20221121 (Red Hat 12.2.1-4) 

```♥️ cat stderr.2
Using Python /usr/bin/python3
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.1.0
built-in source types: file fcd rtl rtl_tcp uhd hackrf rfspace airspy soapy redpitaya 
Using device #0 Realtek RTL2838UHIDIR SN: 14
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
gain: name: LNA range: start 0 stop 49 step 0
setting gain lna to 36
supported sample rates 250000-2560000 step 24000
[R82XX] PLL not locked!
Using two-stage decimator for speed=960000, decim=10/4 if1=96000 if2=24000
Traceback (most recent call last):
  File "/usr/lib64/python3.10/site-packages/gnuradio/gr/hier_block2.py", line 33, in wrapped
    endp = [(p.to_basic_block(), 0) if hasattr(p, 'to_basic_block')
  File "/usr/lib64/python3.10/site-packages/gnuradio/gr/hier_block2.py", line 34, in <listcomp>
    else (p[0].to_basic_block(), p[1]) for p in points]
TypeError: 'gnuradio.op25_repeater.op25_repeater_python.gardner_cc' object is not subscriptable

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/mdulaney/rpmbuild/BUILD/op25-gr310/op25/gr-op25_repeater/apps/./rx.py", line 1101, in <module>
    rx = rx_main()
  File "/home/mdulaney/rpmbuild/BUILD/op25-gr310/op25/gr-op25_repeater/apps/./rx.py", line 1005, in __init__
    self.tb = p25_rx_block(self.options)
  File "/home/mdulaney/rpmbuild/BUILD/op25-gr310/op25/gr-op25_repeater/apps/./rx.py", line 241, in __init__
    self.open_usrp()
  File "/home/mdulaney/rpmbuild/BUILD/op25-gr310/op25/gr-op25_repeater/apps/./rx.py", line 909, in open_usrp
    self.__set_rx_from_osmosdr()
  File "/home/mdulaney/rpmbuild/BUILD/op25-gr310/op25/gr-op25_repeater/apps/./rx.py", line 838, in __set_rx_from_osmosdr
    self.__build_graph(self.src, capture_rate)
  File "/home/mdulaney/rpmbuild/BUILD/op25-gr310/op25/gr-op25_repeater/apps/./rx.py", line 318, in __build_graph
    self.demod = p25_demodulator.p25_demod_cb( msgq_id = 0,
  File "/home/mdulaney/rpmbuild/BUILD/op25-gr310/op25/gr-op25_repeater/apps/p25_demodulator.py", line 430, in __init__
    self.connect_chain(demod_type)
  File "/home/mdulaney/rpmbuild/BUILD/op25-gr310/op25/gr-op25_repeater/apps/p25_demodulator.py", line 504, in connect_chain
    self.connect(self.if_out, self.cutoff, self.agc, self.fll, self.clock, self.diffdec, self.costas, self.to_float, self.rescale, self.slicer)
  File "/usr/lib64/python3.10/site-packages/gnuradio/gr/hier_block2.py", line 36, in wrapped
    raise ValueError("Unable to coerce endpoints: " + str(err))
ValueError: Unable to coerce endpoints: 'gnuradio.op25_repeater.op25_repeater_python.gardner_cc' object is not subscriptable
Reattached kernel driver
```

boatbod, 2023-02-11:
Can you please post the output of "gnuradio-config-info --pybind" and 
also "python3 --version".

The bindings in my gr310 repo are built using pybind-2.9.1 and python3 
ver 3.10.6.
If your RedHat pybind11-dev version is significantly different, or your 
python3 has a different ABI then that could cause the mismatch.  Under 
those circumstances the only fix would be to rebuild the bindings 
locally, but for that you'll also need to implement my gcc-12 gnuradio 
modtool patch.  I'll provide details on that as needed.

mldulaney, 2023-02-11:
```
♥️ python --version
Python 3.10.9
♥️ gnuradio-config-info --pybind
2.9.2
♥️ rpm -q pybind11-devel
pybind11-devel-2.9.2-1.fc36.x86_64
```

What's this gnuradio modtool patch?

boatbod, 2023-02-11:
With those version everything _should_ be compatible...

If you want to experiment, try rebuilding the bindings yourself. Since 
you are using gcc-12 you will probably run into errors when running 
"gr_modtool bind".  This can be worked around with a small change to one 
of the installed gnuradio python files.  As a test, try the following 
commands:

    cd ~/op25/op25/gr-op25
    gr_modtool bind decoder_bf

If it errors with the following (and more) complaints, you need to apply 
a simple patch to a gnuradio source file named 
'parseheader_generic.py'.  The exact location of the file depends on how 
gnuradio was installed, so you'll probably have to hunt it down.

In file included from /usr/include/x86_64-linux-gnu/c++/12/bits/c++allocator.h:33:
/usr/include/c++/12/bits/new_allocator.h:153:6: error: call to'__builtin_operator_delete'  selects
       non-usual deallocation function

Track down the file using "find", assuming either in /usr/local/lib or 
/usr/lib

    find /usr/lib /usr/local/lib -name 'parseheader_generic.py'
    /usr/local/lib/python3.10/dist-packages/gnuradio/blocktool/core/parseheader_generic.py

One you find the file, go ahead and patch it

    sudo patch -b
    /usr/local/lib/python3.10/dist-packages/gnuradio/blocktool/core/parseheader_generic.py
    ~/op25/parseheader_generic.py.patch

Retry the gr_modtool bind test above and this time it should work. 
Assuming it does, go ahead and build all the bindings.

    cd ~/op25
    ./rebuild_bindings.sh

After that, rebuild the whole op25 codebase.  Once the build and install 
finishes, test it in the usual way.

    cd ~/op25
    ./rebuild.sh

mldulaney, 2023-02-11:
```
♥️ gr_modtool bind decoder_bf
GNU Radio module name identified: op25
Writing binding code to python/op25/bindings/decoder_bf_python.cc
Writing binding code to python/op25/bindings/docstrings/decoder_bf_pydoc_template.h
```

I'll try rebuilding pybind

mldulaney, 2023-02-12:
Rebuilt pybind then rebuilt gnuradio and then op25, same result.

boatbod, 2023-02-13:
After considerable pain I now have a running RHEL 9 instance inside a virtual environment. Unless I'm missing something, I don't see a gnuradio package being available via yum?  If you want me to attempt to reproduce/fix your issue I'm going to need details of the steps you took to get gnuradio installed and op25 compiled.  Thanks.

mldulaney, 2023-02-14:
Sorry, it's Fedora 36 I'm running.

I have a copr repository with rpms that you can install; all I did to build gnuradio etc. is rebuild the existing source rpms against the newer gcc.

https://copr.fedorainfracloud.org/coprs/jdulaney/radio/builds/

boatbod, 2023-02-15:
>>Sorry, it's Fedora 36 I'm running.
Well that puts a different spin on things!

So I went back and installed Fedora 37 and am presently building op25.  There were way too many packages that I had to install to even get to the point where op25's cmake didn't barf.  I probably should have written them all down, but at a guess I'm going to say you need to install the following packages to make the bindings work properly:
sudo yum install pybind11 castxml clang-devel
pip install pygccxml

mldulaney, 2023-02-17:
How are you invoking cmake?  Even if I just do cmake .. from the build dir, I get errors (though different ones; modules not able to be imported).  My goal is to make this an RPM package.  The standard invocation of cmake in Fedora is:

/usr/bin/cmake -S . -B redhat-linux-build -DCMAKE_C_FLAGS_RELEASE:STRING=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE:STRING=-DNDEBUG -DCMAKE_Fortran_FLAGS_RELEASE:STRING=-DNDEBUG -DCMAKE_VERBOSE_MAKEFILE:BOOL=ON -DCMAKE_INSTALL_DO_STRIP:BOOL=OFF -DCMAKE_INSTALL_PREFIX:PATH=/usr -DINCLUDE_INSTALL_DIR:PATH=/usr/include -DLIB_INSTALL_DIR:PATH=/usr/lib64 -DSYSCONF_INSTALL_DIR:PATH=/etc -DSHARE_INSTALL_PREFIX:PATH=/usr/share -DLIB_SUFFIX=64 -DBUILD_SHARED_LIBS:BOOL=ON

boatbod, 2023-02-17:
cmake invocation is about as simple as it gets:
```
cd ~/op25
mkdir build
cd build
cmake ../
make
sudo make install

```
If you're still seeing cmake errors you are likely missing one or more packages.
Post the error and I'll offer some guidance.,

mldulaney, 2023-02-22:
I'm not getting errors from cmake; I have all packages installed.  The issue is that when I invoke cmake the way the fedora build system does, I get the error I opened this here issue with.

boatbod, 2023-02-22:
If you can capture logs of a build using the following commands I'd like 
to take a deeper dive.

cd ~/op25
rm -rf build/*
./build_bindings.sh 2>&1 | tee build/bindings.log
cd build
cmake -DCMAKE_INSTALL_PREFIX="/usr" ../ 2>&1 | tee cmake.log
make 2>&1 | tee build.log
sudo make install 2>&1 | tee install.log

This should result in four .log files in the build directory 
(bindings.log, cmake.log, build.log, install.log)

Thanks,
Graham

mldulaney, 2023-03-10:
Hi, so it built just fine following those instructions.  However, I get the error I opened the bug for if I use the full set of cmake flags that the fedora build system uses.

to reproduce, and to modify your above commands:

cd ~/op25
rm -rf build/*
./build_bindings.sh 2>&1 | tee build/bindings.log
cd build
cmake .. -DCMAKE_C_FLAGS_RELEASE:STRING=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE:STRING=-DNDEBUG -DCMAKE_Fortran_FLAGS_RELEASE:STRING=-DNDEBUG -DCMAKE_VERBOSE_MAKEFILE:BOOL=ON -DCMAKE_INSTALL_DO_STRIP:BOOL=OFF -DCMAKE_INSTALL_PREFIX:PATH=/usr -DINCLUDE_INSTALL_DIR:PATH=/usr/include -DLIB_INSTALL_DIR:PATH=/usr/lib64 -DSYSCONF_INSTALL_DIR:PATH=/etc -DSHARE_INSTALL_PREFIX:PATH=/usr/share -DLIB_SUFFIX=64 -DBUILD_SHARED_LIBS:BOOL=ON | tee cmake.log
make 2>&1 | tee build.log
sudo make install 2>&1 | tee install.log

boatbod, 2023-03-10:
If I understand correctly, building and running op25 the way I suggested above works just fine, but when you add all those extra cmake options the code breaks? That being the case, this is not an op25 issue but a platform-specific build configuration for which there is already a solution.  Thanks for helping get to this point - I'm going to close this problem report out now.

closed 2023-03-10 by boatbod

mldulaney, 2023-03-12:
So, basically, what you're saying is that your code breaks in a way that thousands of other projects don't and you don't consider that a problem?

boatbod, 2023-03-12:
I'm saying the code built and ran just fine on my Fedora test environment and you successfully built and ran it on your system using my build instructions.  At this point I have invested all the time that I am prepared to spend.  If that does not sit well with you, feel free to try op25 osmocom or some other software such as sdrtrunk.