boatbod/op25 #184: multi_rx.py doesn't seem to support more than one trunk on a single dongle

semisynthetic, 2023-05-23:
I've been working with multi_rx.py configuration, and I can get things working fine with one P25 trunk and channel configured (for my single rtl-sdr). However, when I try to add a second P25 trunk, it seems it will not allow this.
I get 
* * * Channel 'NCViperPB' cannot share a tunable device - ignoring!
in the log file.
I was assuming this would work somehow based on these notes in the README.
"Multi-system/multi-channel concurrent operation (full time, not sequential)
Single, Multiple and Shared SDR devices (e.g. wideband devices such as Airspy etc)"

Also, I was assuming I should at least be able to replicate the functionality of rx.py, on some level, and I can't seem to do that.

Here's the config I'm working with (I'm outputting audio to liquidsoap, or at least trying to).

```
{
	"channels": [
		{
			"name": "AshevilleP25PB",
			"device": "sdr0",
			"trunking_sysname": "AshevilleP25",
			"demod_type": "cqpsk",
			"cqpsk_tracking": true,
			"tracking_threshold": 120,
			"tracking_feedback": 0.75,
			"destination": "udp://127.0.0.1:23456",
			"excess_bw": 0.2,
			"filter_type": "rc",
			"if_rate": 24000,
			"plot": "",
			"symbol_rate": 4800,
			"enable_analog": "off",
			"blacklist": "",
			"whitelist": ""
		},
		{
			"name": "NCViperPB",
			"device": "sdr0",
			"trunking_sysname": "NCViper",
			"demod_type": "cqpsk",
			"cqpsk_tracking": true,
			"tracking_threshold": 120,
			"tracking_feedback": 0.75,
			"destination": "udp://127.0.0.1:23466",
			"excess_bw": 0.2,
			"filter_type": "rc",
			"if_rate": 24000,
			"plot": "",
			"symbol_rate": 4800,
			"enable_analog": "off",
			"blacklist": "",
			"whitelist": ""
		}
	],
	"devices": [
		{
			"args": "rtl=0",
			"gains": "LNA:47",
			"gain_mode": false,
			"name": "sdr0",
			"offset": 0,
			"ppm": 0.0,
			"rate": 2400000,
			"usable_bw_pct": 0.85,
			"tunable": true
		}
	],
	"trunking": {
		"module": "tk_p25.py",
		"chans": [
			{
				"nac": "0x0",
				"sysname": "AshevilleP25",
				"control_channel_list": "852.6875,852.9625,853.825",
				"whitelist": "",
				"blacklist": "",
				"tgid_tags_file": "ashevillep25.tsv",
				"tdma_cc": false,
				"crypt_behavior": 2
			},
			{
				"nac": "0x0",
				"sysname": "NCViper",
				"control_channel_list": "852.6875,852.9625,853.825,853.7625,858.7875,855.2125",
				"whitelist": "",
				"blacklist": "",
				"tgid_tags_file": "ncviper.tsv",
				"tdma_cc": false,
				"crypt_behavior": 2
			}
		]
	},
	"audio": {
		"module": "sockaudio.py",
		"instances": [
			{
				"instance_name": "",
				"device_name": "pulse",
				"udp_port": 23426,
				"audio_gain": 1.0,
				"number_channels": 1
			}
		]
	},
	"terminal": {
		"module": "terminal.py",
		"#terminal_type": "http:127.0.0.1:8080",
		"terminal_type": "curses",
		"curses_plot_interval": 0.1,
		"http_plot_interval": 1.0,
		"http_plot_directory": "../www/images",
		"tuning_step_large": 1200,
		"tuning_step_small": 100
	}
}
```

Any idea what I'm doing wrong?

boatbod, 2023-05-24:
Multi_rx supports two different methods of monitoring simultaneous voice channels.  The key thing to remember is that every individual channel has to have access to the full RF spectrum of the P25 system being monitored without affecting any of the other channels.  To that end, since most P25 systems occupy more than the ~2Mhz max usable bandwidth available from the cheap RTL dongles, you either need to dedicate one dongle per voice channel and make the devices tunable (tunable=true) or have one or more wideband receivers (e.g. Airspy) set up with sufficient bandwidth to capture the whole P25 system at the same time and designate it non-tunable (tunable=false).  What you can't do is share a tunable device between two channels, because re-tuning channel A would implicitly affect channel B.

In addition to the above, multi_rx **does not** support the system hopping capability that could be achieved in rx.py when multiple lines were entered into trunk.tsv.  My reasoning is that RTL devices are inexpensive (~$25) and the advantage of true simultaneous monitoring far outweighs the disadvantages of the small added cost.  With rx.py the time-based hopping between systems just caused lots of traffic to be missed.

semisynthetic, 2023-05-24:
OK, understood, and I agree. The time it takes for these freqs to monitor in rx.py, a lot is lost. Thanks!

closed 2023-05-24 by semisynthetic