boatbod/op25 #186: Seg Fault on Ubuntu 22.04.2

Davem734, 2023-07-03:
Using:
  Ubuntu 22.04.2 LTS
  Tried both RTL-SDR and SZ-PLZ radios and got the seg fault
  Both radios work with GQRX and can see the signal at 855.290Mhz

Installed using:
```
  sudo apt -y install gnuplot-x11  
  mkdir -p ~/Apps/OP25   
  cd ~/Apps/OP25
  git clone https://github.com/boatbod/op25.git
  cd ~/Apps/OP25/op25
  git checkout gr310
  ./install.sh 
  cd ~/Apps/OP25/op25/op25/gr-op25_repeater/apps
```
Attempted to run using:

`  ~/Apps/OP25/op25/op25/gr-op25_repeater/apps$ ./rx.py --args 'SDRplay' -S 8000000 -f 855.290e6 -2 --tdma-cc -O pulse -v 11
`

Results:
```
  Using Python /usr/bin/python3
  gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.1.1
  built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
  [INFO] [UHD] linux; GNU C++ version 11.2.0; Boost_107400; UHD_4.1.0.5-3
  libusb: warning [libusb_exit] device 2.2 still referenced
  libusb: warning [libusb_exit] device 2.1 still referenced
  libusb: warning [libusb_exit] device 1.4 still referenced
  libusb: warning [libusb_exit] device 1.3 still referenced
  libusb: warning [libusb_exit] device 1.5 still referenced
  libusb: warning [libusb_exit] device 1.1 still referenced
  -- Using subdev spec '0:0'.
  supported sample rates 8000-192000 step 3900
  Using two-stage decimator for speed=8000000, decim=80/4 if1=100000 if2=25000
  p25_frame_assembler_impl: do_imbe[0], do_output[0], do_audio_output[0], do_phase2_tdma[1], do_nocrypt[0]
  07/03/23 10:02:54.626748 [0] p25_frame_assembler_impl::control: cmd(set_slotid), args({"tuner": 0, "cmd": "set_slotid", "slotid": 0})
  Segmentation fault (core dumped)
```
One complicating factor could be the fact that I tried the osmocom.org version first. That also did not work. I removed it using the following commands befor installing the boatbod version:
```
 $ sudo find /usr/local -name '*op25*' -exec rm -rf {} \;
 $ sudo ldconfig
```
Not sure how to begin troubleshooting this.
Any help would be appreciated.

boatbod, 2023-07-03:
Dave, The segv is almost certainly being caused by building while there were bits of Osmocom op25 hanging around.   Recommend you uninstall op25 from the build directory, delete the build directory contents,  then check the file system for anything "*op25*" and delete whatever you find.   Finally, go back through the op25 build process and it should be clean. Graham

Davem734, 2023-07-04:
I followed your instructions. The uninstall process reported some errors
and did not remove all of the op25 files so I removed them manually. 

The install script only reported warnings as far as I can tell.
Attached is the full output of the install script in case it is needed. 

The results are interesting. The first time I ran rx.py I got some 
errors reported ending with 'MemoryError: std::bad_alloc'. 

The second time I ran rx.py I got no errors reported
just the 'Segmentation fault (core dumped)' message.
After that, I only get the 'Segmentation fault (core dumped)' message
even after a power down and reboot.

```
$ cd ~/Apps/OP25/op25/build

$ sudo make uninstall
  [sudo] password for davem: 
  -- Uninstalling "/usr/local/lib/cmake/gnuradio-op25/gnuradio-op25Config.cmake"
  .
  .
  .
  -- Uninstalling "/usr/local/share/gnuradio/grc/blocks/op25_repeater_fsk4_slicer_fb.xml"
  Built target uninstall
  CMake Error at cmake_uninstall.cmake:4 (MESSAGE):
    Cannot find install manifest:
    "/home/davem/Apps/OP25/op25/build/op25/gr-op25/install_manifest.txt"
  make[3]: *** [op25/gr-op25/CMakeFiles/uninstall.dir/build.make:70: op25/gr-op25/CMakeFiles/uninstall] Error 1
  make[2]: *** [CMakeFiles/Makefile2:382: op25/gr-op25/CMakeFiles/uninstall.dir/all] Error 2
  make[1]: *** [CMakeFiles/Makefile2:389: op25/gr-op25/CMakeFiles/uninstall.dir/rule] Error 2
  make: *** [Makefile:179: uninstall] Error 2

$ sudo sudo rm -rf ~/Apps/OP25/op25

$ sudo find /usr/local -name '*op25*'
  /usr/local/include/gnuradio/op25
  /usr/local/include/gnuradio/op25_repeater
  /usr/local/lib/python3.10/dist-packages/gnuradio/op25
  /usr/local/lib/python3.10/dist-packages/gnuradio/op25_repeater
  /usr/local/lib/cmake/gnuradio-op25_repeater
  /usr/local/lib/cmake/gnuradio-op25
$ sudo find /usr/local -name '*op25*' -exec rm -rf {} \; 
$ sudo find /usr/local -name '*op25*'

$ git clone https://github.com/boatbod/op25.git
$ cd ~/Apps/OP25/op25
$ git checkout gr310
$ ./install.sh

$ cd ~/Apps/OP25/op25/op25/gr-op25_repeater/apps
$ ./rx.py --args 'SDRplay' -S 8000000 -f 855.290e6 -2 --tdma-cc -O pulse -v 11
  Using Python /usr/bin/python3
  gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.1.1
  built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
  [INFO] [UHD] linux; GNU C++ version 11.2.0; Boost_107400; UHD_4.1.0.5-3
  libusb: warning [libusb_exit] device 2.2 still referenced
  libusb: warning [libusb_exit] device 2.1 still referenced
  libusb: warning [libusb_exit] device 1.4 still referenced
  libusb: warning [libusb_exit] device 1.3 still referenced
  libusb: warning [libusb_exit] device 1.5 still referenced
  libusb: warning [libusb_exit] device 1.1 still referenced
  -- Using subdev spec '0:0'.
  supported sample rates 8000-192000 step 3900
  Using two-stage decimator for speed=8000000, decim=80/4 if1=100000 if2=25000
  p25_frame_assembler_impl: do_imbe[0], do_output[0], do_audio_output[0], do_phase2_tdma[1], do_nocrypt[0]
  07/04/23 12:31:09.307701 [0] p25_frame_assembler_impl::control: cmd(set_slotid), args({"tuner": 0, "cmd": "set_slotid", "slotid": 0})
  Traceback (most recent call last):
    File "/home/davem/Apps/OP25/op25/op25/gr-op25_repeater/apps/./rx.py", line 1100, in <module>
      rx = rx_main()
    File "/home/davem/Apps/OP25/op25/op25/gr-op25_repeater/apps/./rx.py", line 1004, in __init__
      self.tb = p25_rx_block(self.options)
    File "/home/davem/Apps/OP25/op25/op25/gr-op25_repeater/apps/./rx.py", line 241, in __init__
      self.open_usrp()
    File "/home/davem/Apps/OP25/op25/op25/gr-op25_repeater/apps/./rx.py", line 908, in open_usrp
      self.__set_rx_from_osmosdr()
    File "/home/davem/Apps/OP25/op25/op25/gr-op25_repeater/apps/./rx.py", line 837, in __set_rx_from_osmosdr
      self.__build_graph(self.src, capture_rate)
    File "/home/davem/Apps/OP25/op25/op25/gr-op25_repeater/apps/./rx.py", line 343, in __build_graph
      self.connect(source, self.demod, self.decoder)
    File "/usr/lib/python3/dist-packages/gnuradio/gr/hier_block2.py", line 39, in wrapped
      func(self, src, src_port, dst, dst_port)
    File "/usr/lib/python3/dist-packages/gnuradio/gr/hier_block2.py", line 106, in connect
      self.primitive_connect(*args)
  MemoryError: std::bad_alloc

$ ./rx.py --args 'SDRplay' -S 8000000 -f 855.290e6 -2 --tdma-cc -O pulse -v 11
  Using Python /usr/bin/python3
  gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.1.1
  built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
  [INFO] [UHD] linux; GNU C++ version 11.2.0; Boost_107400; UHD_4.1.0.5-3
  libusb: warning [libusb_exit] device 2.2 still referenced
  libusb: warning [libusb_exit] device 2.1 still referenced
  libusb: warning [libusb_exit] device 1.4 still referenced
  libusb: warning [libusb_exit] device 1.3 still referenced
  libusb: warning [libusb_exit] device 1.5 still referenced
  libusb: warning [libusb_exit] device 1.1 still referenced
  -- Using subdev spec '0:0'.
  supported sample rates 8000-192000 step 3900
  Using two-stage decimator for speed=8000000, decim=80/4 if1=100000 if2=25000
  p25_frame_assembler_impl: do_imbe[0], do_output[0], do_audio_output[0], do_phase2_tdma[1], do_nocrypt[0]
  07/04/23 12:42:41.131839 [0] p25_frame_assembler_impl::control: cmd(set_slotid), args({"tuner": 0, "cmd": "set_slotid", "slotid": 0})
  Segmentation fault (core dumped)
```
[InstallOutput.txt](https://github.com/boatbod/op25/files/11951439/InstallOutput.txt)

boatbod, 2023-07-05:
Dave,

The initial crash looks to have occurred inside the _build_graph() call, 
which is where gnuradio tries to assemble the flowgraph by joining the 
various blocks together.  Are you using vanilla distro versions of 
gr-osmosdr, librtlsdr and gnuradio?  If you have either local versions 
of these packages, or older versions left behind by a distro upgrade you 
might need to either rebuild them or clean up and ensure only the 
current distro versions exist.

Graham

Davem734, 2023-07-05:
Thank you for your help.

gr-osmosdr Version: 0.2.3-5build2 and librtlsdr0 Version: 0.6.0-4 are from the repo: 
`  http://ca.archive.ubuntu.com/ubuntu jammy/universe amd64 Packages
`
gnuradio Version: 3.10.5.1-0~gnuradio~jammy-2 is from the repo:
 ` https://ppa.launchpadcontent.net/gnuradio/gnuradio-releases/ubuntu jammy/main amd64 Packages
`  
See attached for full details.

[gnuradio-ver.txt](https://github.com/boatbod/op25/files/11962456/gnuradio-ver.txt)

Davem734, 2023-07-06:
Removed gnuradio Version: 3.10.5.1-0~gnuradio~jammy-2
Installed gnuradio Version: 3.10.1.1-2
Removed and reinstalled op25

$ ./rx.py --args 'SDRplay' -S 8000000 -f 855.290e6 -2 --tdma-cc -O pulse -v 11 2> stderr.2

Does not seg fault. So this seems to have fixed that problem.

But it shows this error at the top of the display:
`thread_body_wrapper :error: ERROR thread[thread-per-block[1]: <block usrp_source(3)>]: RtAudio init error 'RtApiPulse::probeDeviceOpen: error connecting input to PulseAudio server.
`
```
$ less stderr.2
  Using Python /usr/bin/python3
  gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.1.1
  built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
  [INFO] [UHD] linux; GNU C++ version 11.2.0; Boost_107400; UHD_4.1.0.5-3
  libusb: warning [libusb_exit] device 2.2 still referenced
  libusb: warning [libusb_exit] device 2.1 still referenced
  libusb: warning [libusb_exit] device 1.4 still referenced
  libusb: warning [libusb_exit] device 1.3 still referenced
  libusb: warning [libusb_exit] device 1.5 still referenced
  libusb: warning [libusb_exit] device 1.1 still referenced
  -- Using subdev spec '0:0'.
  supported sample rates 8000-192000 step 3900
  Using two-stage decimator for speed=8000000, decim=80/4 if1=100000 if2=25000
  p25_frame_assembler_impl: do_imbe[0], do_output[0], do_audio_output[0], do_phase2_tdma[1], do_nocrypt[0]
  07/05/23 20:40:46.942978 [0] p25_frame_assembler_impl::control: cmd(set_slotid), args({"tuner": 0, "cmd": "set_slotid", "slotid": 0})
  metadata update not enabled
  python version detected: 3.10.6 (main, May 29 2023, 11:10:38) [GCC 11.3.0]
  [INFO] [UHDSoapyDevice] Using format CF32.

```
Not sure if it is working or not but the seg fault problem is fixed. 
Thank you for your help and for all of the work you have put into maintaining the op25 project...

boatbod, 2023-07-08:
Pulse Audio is rather quirky when it comes to permissions, especially if 
you're running a headless environment where the desktop doesn't get 
started at boot time.  Best I can suggest is switching over to ALSA 
"default" audio device rather than "pulse".

Davem734, 2023-07-11:
I have tried using the following:

```
-O pulse
$ ./rx.py --args 'SDRplay' -S 8000000 -f 855.290e6 -2 --tdma-cc -D cqpsk -O pulse -v 11 2> stderr.2

-U 
$ ./rx.py --args 'SDRplay' -S 1000000 -f 855.290e6 -q 0 --tdma-cc -2 -D cqpsk -U -v 11 2> stderr.2

-O plughw:0,0
$ ./rx.py --args 'SDRplay' -S 1000000 -f 855.290e6 -q 0 --tdma-cc -2 -D cqpsk -O plughw:0,0 -v 11 2> stderr.2

-O ALSA 
$ ./rx.py --args 'SDRplay' -S 1000000 -f 855.290e6 -q 0 --tdma-cc -2 -D cqpsk -O ALSA -v 11 2> stderr.2
```
All show the me the same error message overwritten on the op25 display:
`  thread_body_wrapper :error: ERROR thread[thread-per-block[1]: <block usrp_source(3)>]: RtAudio init error 'RtApiPulse::probeDeviceOpen: error connecting input to PulseAudio server.
`
Pressing 'q' will close the display but I have to kill the python3 
process to get back to the prompt in the terminal.

What is the preferred command for specifying ALSA output?

The following command works so I know that the sound subsystem is working and accessible:
`  $ aplay -D plughw:0,0 /usr/share/sounds/alsa/Front_Center.wav
`
```
$ less stderr.2 looks like:
  Using Python /usr/bin/python3
  gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.1.1
  built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
  [INFO] [UHD] linux; GNU C++ version 11.2.0; Boost_107400; UHD_4.1.0.5-3
  libusb: warning [libusb_exit] device 2.2 still referenced
  libusb: warning [libusb_exit] device 2.1 still referenced
  libusb: warning [libusb_exit] device 1.4 still referenced
  libusb: warning [libusb_exit] device 1.3 still referenced
  libusb: warning [libusb_exit] device 1.5 still referenced
  libusb: warning [libusb_exit] device 1.1 still referenced
  -- Using subdev spec '0:0'.
  supported sample rates 8000-192000 step 3900
  Using two-stage decimator for speed=1000000, decim=10/4 if1=100000 if2=25000
  p25_frame_assembler_impl: do_imbe[0], do_output[0], do_audio_output[0], do_phase2_tdma[1], do_nocrypt[0]
  07/11/23 17:29:01.163595 [0] p25_frame_assembler_impl::control: cmd(set_slotid), args({"tuner": 0, "cmd": "set_slotid", "slotid": 0})
  metadata update not enabled
  python version detected: 3.10.6 (main, May 29 2023, 11:10:38) [GCC 11.3.0]
  [INFO] [UHDSoapyDevice] Using format CF32.
  stderr.2 (END)
```
Attached is some additional detail regarding my pulseaudio config.

[PulseAudio.txt](https://github.com/boatbod/op25/files/12021907/PulseAudio.txt)

closed 2023-10-28 by boatbod

macalikm, 2024-01-16:
Ive just installed op25 with the gr310 branch on an AMD PC running Ubuntu 22.04
when I run the op25.sh script from the ~/op25/op25/gr-op25_repeater/apps dir, I get:

./op25.sh: line 1: 19175 Bus error               (core dumped) ./rx.py --nocrypt --args "rtl" --gains 'lna:36' -S 960000 -T trunk.tsv 2> stderr.2

./rx.py by itself will start. When I add./rx.py --args "rtl" it bombs 

The sdr is a nooelec RTL-SDR which works just fine with things like readsb and gqrx.
 Running "./rx.py --args 'rtl' -N 'LNA:47' -s 2500000 -x 2 -f 151.010e6 -o 17e3 -q 2" from the command line gives a little more output, but also Bus errors out:

Using Python /usr/bin/python3
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.1.1
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
gain: name: LNA range: start 0 stop 49 step 0
setting gain LNA to 47
supported sample rates 250000-2560000 step 24000
[R82XX] PLL not locked!
demodulator: xlator if_rate=24000, input_rate=960000, decim=40, taps=193, resampled_rate=24000, sps=5
p25_frame_assembler_impl: do_imbe[0], do_output[0], do_audio_output[0], do_phase2_tdma[0], do_nocrypt[0]
Bus error (core dumped)

Can you help?

boatbod, 2024-01-16:
Most probably there's a problem between the versions of gnuradio, 
librtlsdr & gr-osmosdr drivers that op25 was built with, versus the 
versions of libraries found during runtime.  I've seen this happen when 
attempting to use a mix of locally built libraries alongside apt packages.

I'd suggest starting by uninstalling op25 (sudo make uninstall), purge 
anything op25 related from /usr/local/lib, purge the ~op25/build 
directory and the re-build from the beginning.

If you want to actually debug the core and find out what's going on, go 
back and rebuild op25 with full debug symbols and attempt to run rx.py 
under a debugger to find out where it is actually failing.  It can be 
done but is a royal PITA and should be considered a last resort.

Graham

macalikm, 2024-01-18:
Thanks Graham, 
That sounds like exactly what I did, "...use a mix of locally built libraries alongside apt packages." Ill try an uninstall, purge and rebuild this weekend and post the results (if any). Hopefully It wont come to debugging the core. That really isn't one of my strengths. If it goes that fat, I'll probably just start over with a clean Ubuntu 22.04.. Maybe on a Raspi4.
Thanks again,
Mike