boatbod/op25 #187: Taglist, Whitelist, and Blacklist files not found if placed in directory with capital letters.

lroberts-lly, 2023-07-08:
**Describe the bug**

I'm a new user and came across this issue while trying to organize my files.

If the taglist, whitelist, or blacklist files are placed in a directory that contains capital letter(s), the rx.py program will report file not found.

**To Reproduce**

1.  cd to op25-boatbod/op25/gr-op25_repeater/apps
2.  mkdir Configs
3.  place a known good taglist.tsv file in the Configs directory
4.  create a trunking.tsv with an entry for"TGID Tags File" set to ./Configs/taglist.tsv.
5. Set other fields in trunking.tsv to known good values.
6. Execute  ./rx.py --args 'rtl' -N 'LNA:47' -S 2500000 -x 2 -f 854.4375e6 -o 17e3 -X -T ./trunking.tsv -V -2 -U -l 'http:0.0.0.0:8080'

Here is the error output:
Using Python /usr/bin/python3
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
gain: name: LNA range: start 0 stop 0 step 0
setting gain LNA to 47
supported sample rates 250000-2560000 step 24000
Exact sample rate is: 2500000.107620 Hz
[R82XX] PLL not locked!
Using two-stage decimator for speed=2500000, decim=25/4 if1=100000 if2=25000
op25_audio::open_socket(): enabled udp host(127.0.0.1), wireshark(23456), audio(23456)
p25_frame_assembler_impl: do_imbe[1], do_output[0], do_audio_output[1], do_phase2_tdma[1], do_nocrypt[0]
Traceback (most recent call last):
  File "./rx.py", line 1099, in <module>
    rx = rx_main()
  File "./rx.py", line 1003, in __init__
    self.tb = p25_rx_block(self.options)
  File "./rx.py", line 241, in __init__
    self.open_usrp()
  File "./rx.py", line 907, in open_usrp
    self.__set_rx_from_osmosdr()
  File "./rx.py", line 836, in __set_rx_from_osmosdr
    self.__build_graph(self.src, capture_rate)
  File "./rx.py", line 376, in __build_graph
    self.trunk_rx = trunking.rx_ctl(frequency_set = self.change_freq, fa_ctrl = self.control, debug = self.options.verbosity, conf_file = self.options.trunk_conf_file, logfile_workers=logfile_workers, meta_update = self.meta_update, crypt_behavior = self.options.crypt_behavior)
  File "/home/larry/Projects/rtlsdr/op25-boatbod/op25/gr-op25_repeater/apps/trunking.py", line 1276, in __init__
    self.build_config_tsv(conf_file)
  File "/home/larry/Projects/rtlsdr/op25-boatbod/op25/gr-op25_repeater/apps/trunking.py", line 1381, in build_config_tsv
    self.setup_config(configs)
  File "/home/larry/Projects/rtlsdr/op25-boatbod/op25/gr-op25_repeater/apps/trunking.py", line 1454, in setup_config
    with open(configs[nac]['tgid_tags_file'], 'r') as csvfile:
FileNotFoundError: [Errno 2] No such file or directory: './configs/taglist.tsv'

########################################^^^^^^ Notice the "configs" in the last line of the error is lowercase

Digging into the source code I found that op25-boatbod/op25/gr-op25_repeater/apps/helper_funcs.py has instructions to lowercase everything in the trunking.tsv file except the sysname.  I can understand needing this for the hex fields (nac), but not anything else.

    151        for i in range(len(row)):
    152                 if row[i]:
    153                     fields[hdrmap[i]] = row[i]
    154                     if hdrmap[i] != 'sysname':
    156                           fields[hdrmap[i]] = fields[hdrmap[i]].lower()

I was able to resolve this problem by changing line 154 to:

   154                      if hdrmap[i] not in ['sysname','tgid_tags_file','whitelist','blacklist'] :

boatbod, 2023-07-08:
Thanks for the bug report.  I changed the code so that numeric fields are forced lowercase, but alpha fields are not.
e.g. a hex nac has to be lowercase, and a frequency containing an "e" also has to be lowercase.

closed 2023-07-08 by boatbod

lroberts-lly, 2023-07-09:
Thanks for the quick turn around!!