boatbod/op25 #189: Errors with Latest Python Version

krutzy62, 2023-08-05:
**Describe the bug**
When upgrading the software on Linux Mint 21.1 machine, OP25 crashes with Python issues.  

python version detected: 3.10.6 (main, Nov 14 2022, 16:10:14) [GCC 11.3.0]
post_req: error processing input: b''
Traceback (most recent call last):
File "/home/ku/op25/op25/gr-op25_repeater/apps/http_server.py", line 71, in post_req
data = json.loads(postdata)
File "/usr/lib/python3.10/json/__init__.py", line 346, in loads
return _default_decoder.decode(s)
File "/usr/lib/python3.10/json/decoder.py", line 337, in decode
obj, end = self.raw_decode(s, idx=_w(s, 0).end())
File "/usr/lib/python3.10/json/decoder.py", line 355, in raw_decode
raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

If I go to rebuild OP25 the complier fails with translator error:
"lto-wrapper: warning: using serial compilation of 5 LTRANS jobs
during RTL pass: final
/usr/include/pybind11/detail/internals.h: In function ‘translate_exception’:
/usr/include/pybind11/detail/internals.h:386:1: internal compiler error: in ix86_print_operand_address_as, at config/i386/i386.c:13598
386 | }
| ^
0x7f207ef37d8f __libc_start_call_main
../sysdeps/nptl/libc_start_call_main.h:58
0x7f207ef37e3f __libc_start_main_impl
../csu/libc-start.c:392
Please submit a full bug report,
with preprocessed source if appropriate.
Please include the complete backtrace with any bug report.
See <file:///usr/share/doc/gcc-11/README.Bugs> for instructions.

**To Reproduce**
Run OP25 or rebuild.

**Screenshots**
N/A

**System Information (please complete the following information):**
 - OS: Linux Mint 21.1
 - GNURadio version 3.10.1.1 - most likely 
 - GCC version 11.3.0

**Additional context**
After I saw the errors I used timeshift to go back to the last known good version.   I may not have all the details wanted but I can redo if necessary.

boatbod, 2023-08-05:
I'm not quite sure what you've got going on, but I've got doubts that it's related to the python version, or gnuradio version.
Since I don't presently have a Linux Mint 21.1 system configured in my virtual environment, I tried a clean build using Ubuntu 22.10 which has python 3.10.7, gnuradio 3.10.3 and gcc 12.2.0 and it compiles, installs and runs with no issue.

If I can find the time I'll install a Mint 21.1, but that's just a Ubuntu derivative so it shouldn't behave an differently.

krutzy62, 2023-08-05:
I am willing to reproduce - but before I do are there specific items I can check and post so I can restore and get back up.  I can give you the exact information that way.

boatbod, 2023-08-05:
You could attempt to reproduce starting from scratch with a clean 
system, but I'm actually wondering if there is something in your config 
that's making the json decode fail.  When I've seen errors like that 
before it's been related to Unicode vs ASCII; in other words something 
8bit got in where we were only expecting 7 bit.

Graham

krutzy62, 2023-08-06:
Let me try it again and get some details.  I did see a new version  of GCC came out 7/27/2023.  Not sure if the update pulled that in or not.
I don't think you suggestion would help.  I have done a number of "git pull" many times on this PC and if it were what you said I would have run into the problem earlier.  Not sure how the config would get screwed up, this micro-PC is dedicated to OP25 and I do nothing else on.  It just runs OP25 24/7 using two RTL-SDR's.  
Let me dig a little.

krutzy62, 2023-08-06:
Actually I think that might be it.  The only way I can show  is these attachments.  

It wants to update both gcc-11 and gcc-12.  I am not programmer in this world so I don't know what gcc-11 and gcc-12 are.
It is also updating Python  the Python Standard Library Extension.
[image]
[image]

boatbod, 2023-08-06:
> image

gcc (and it's sibling gxx) are the C and C++ compiler suite.  Most 
problems found in pybind11 are usually traced to changes in the compiler.

Graham

krutzy62, 2023-08-06:
So I guess I just won't update for a while.  I don't know enough to be a helpful to figuring out how to get around it.  It looks like a change in the syntax to my old eyes.  I am very sorry can't help more.

boatbod, 2023-08-06:
I set up a fresh Mint 21.1 system, then built and ran op25 from the gr310 branch with no issue. After that I pulled all the available software updates, deleted the op25/build directory and rebuilt from scratch. Once again it compiled, installed and ran just fine.

Not sure what to recommend other than you apparently have some inconsistency in either your build or your base os.  It's probably easier if you just nuke it and start over.

krutzy62, 2023-08-06:
Thanks!  I will try later this week when I have time to do it right.

krutzy62, 2023-08-06:
Excuse my ignorant question.   But let me give it a go.  If I were to delete the build directory then do a "git pull" it will pull everything down so I can rebuild everything.  I could see how doing this could pickup new library's so it would compile correctly vs. just updating the new code.  What I am trying to avoid is having to backup all my stuff, reinstall Mint, and then doing op25, liquidsoap and icecast2 again.
If I could just kill those processes, and rebuild OP25 using the above, and restarting then that wouldn't be a big deal.

boatbod, 2023-08-06:
Try this:cd ~/op25/buildsudo make uninstall  (this will error, but do it anyway) cd ~/op25rm -rf build/*./install.shGraham

krutzy62, 2023-08-07:
I will try that tomorrow.  That makes total sense.  Thank you.

boatbod, 2023-08-07:
The other thing you might need to do is make sure you haven't 
inadvertently attempted to regenerate the bindings as I've found 
pybind11 to be rather temperamental at best.  After nuking "build" but 
before you re-runing "install.sh" try this:

cd ~/op25
git checkout .

G.

krutzy62, 2023-08-07:
Update:  Your instructions worked like a champ.  Compiled clean and running well.
I have save these instructions for future reference.
Thank you very much!

closed 2023-08-07 by krutzy62

boatbod, 2023-08-07:
There's a script that's part of the op25 distro which does most of what 
I suggested.  It's called "rebuild.sh"

G.