boatbod/op25 #199: Blog V4 compatibility issue?

theghostglitch, 2024-01-18:
**Describe the bug**
Using a Blog V4. No matter what frequency is selected frequency peaks stay the same.

**To Reproduce**
Steps to reproduce the behavior:
Run OP25 with a Blog v4

**Screenshots**
[Image1](https://drive.google.com/file/d/1C85ngm06OD632WJZKJAiWTaR5neZhfRO/view?usp=drive_link)
[Image2](https://drive.google.com/file/d/19EFtY6oNq6S_6HLZ2nI-0ecjaNUQvnHc/view?usp=drive_link)

**System Information (please complete the following information):**
 - OS: Ubuntu 22.04.3 & Raspbian on Raspberry Pi 3
 - GNURadio version: 3.10
 - GCC version: 11.4.0

**Additional context**
The Blog v4 needed a driver update over the original RTL-SDR driver. It's compatible with the older SDRs but has added features that are required to run the v4 correctly. Not sure if this has been implemented on OP25?

boatbod, 2024-01-19:
> Update
> Installing the v4 drivers manually breaks OP25. Specifically osmosdr
> with an error "osmosdr sorce_c creation failure".
> ./install.sh and a reboot does not fix this.

OP25 relies on gr-osmosdr for interface to the RTL device.  If there has 
been an underlying API change between osmo and rtl to support v4 
drivers, you're probably going to need an update gr-osmosdr as well.

Supplying some debug logs at "-v 11" would be helpful if you'd like me 
to look at this in any more detail.

Thanks,
Graham

boatbod, 2024-01-28:
No futher debug info provided by reporting party.
Suspect the manual installation of blog v4 drivers without installing a compatible version of gr-osmosdr and then rebuilding op25 caused a library incompatibility.

closed 2024-01-28 by boatbod