boatbod/op25 #200: Smartnet & cqpsk vs fsk4

rfuchs, 2024-01-19:
Hi,

Is there a reason why the smartnet NB FM demodulator is only enabled for fsk4 channels but not for cqpsk channels? See https://github.com/boatbod/op25/blob/master/op25/gr-op25_repeater/apps/p25_demodulator_dev.py#L531

I'm listening to https://www.radioreference.com/db/sid/861 and with fsk4 I get distorted digital audio but good FM, and with cqpsk I get good digital audio but no FM. Patching up that line to also allow cqpsk makes things work.

Is this an oversight or have I configured something wrong?

boatbod, 2024-01-21:
The proper voice channel setup for Smartnet can be found in 
smartnet_example.json.
Specifically the following parameters:
     demod_type="fsk4"
     filter_type="widepulse"
If that's not getting you good digital audio, try changing 
filter_type="rc" and see if that helps.

I'm actually a little surprised that demod_type="cqpsk" works at all for 
digital audio since it is supposed to follow the P25 Phase 1 specs which 
mandate C4FM/FSK4 encoding.  The problem is, I've never seen an actual 
Motorola spec, just TIA-102 standards docs.

Let me know whether changing widepulse makes any difference.
Graham

rfuchs, 2024-01-21:
Yeah I tried with the default settings (fsk4/widepulse) first and that gave me very choppy audio. For some reason changing to cqpsk made it work. BUT fsk4/rc also seems to work perfectly. Solved, thanks!

closed 2024-01-21 by rfuchs

boatbod, 2024-01-21:
Do you know if your Smartnet system is simulcast?  Widepulse (for simulcast) increases the filter bandwidth vs the rc filter used by standard P25. Graham

rfuchs, 2024-01-21:
> Do you know if your Smartnet system is simulcast?  Widepulse (for simulcast) increases the filter bandwidth vs the rc filter used by standard P25.

No idea. Is there a way I can tell? Otherwise I only know what's listed on RR.