boatbod/op25 #219: Adding multiple control channels in trunk.tsv file?

k5map, 2024-06-02:
The P25 Phase 2 trunk system I want to monitor has 2 Control frequencies.  Can I add them with comma separated in the trunk.tsv file?  If not, is there way to configure OP25 to use both channels?

foxracr10, 2024-06-02:
Yes that is correct.  I monitor my local system with 9 control channels
that way.

boatbod, 2024-06-03:
Yes and no.  In the rx.py variant of the app, you can list multiple lines in the trunk.tsv file and this will cause the app to cycle between p25 systems (each required to have a unique NAC) and once traffic is found according to the options whitelists/blacklists it will stop and play.  The problem is it takes time to cycle around the control channels (each re-tune takes ~750ms) so your probability of calls being cut off are high, along with there being no way to prioritize between tgids.

The multi_rx.py (newer) version of the app approaches the concept differently; you can define multiple p25 systems - each with it's own control channel - and have them monitored and decoded simultaneously.  Sounds great, but there's no free lunch because the cost of this additional capability is either the need to dedicate one RTL dongle per system, or use of a wide-band sdr with a sample range covering the full spectrum of all the systems needing to be monitored.

To the op: one thing I'm not clear on, is whether your multiple control channels have different NACs and are thus separate p25 systems, or somehow they are being used for load sharing within a single system.  (i.e. two active primary control channels, both carrying voice grants for the full range of tgids).  If the latter, I'm not aware of how such capability is formally supported within TIA-102.

k5map, 2024-06-03:
The 2 control channels are for the same P25 system.  I thought I read a post somewhere indicating I can put both with a comma separating them in the trunk.tsv file but wanted to confirm with anyone who has this working.

wgbecks, 2024-06-03:
P25 TRS are often composed of multiple sites, whereby each site is assigned a NAC, and primary control channel along with one or more alternate control channels.  In the case of monitoring

a single site withing a multisite system, you do configure a comma separated list of primary and alternates in trunk.tsv that is used the older rx.py application.  This list serves to maintain a

continouity of

wgbecks, 2024-06-03:
P25 TRS are often composed of multiple sites, whereby each site is assigned a NAC, and primary control channel along with one or more alternate control channels.  In the case of monitoring

a single site withing a multisite system, you do configure a comma separated list of primary and alternates in trunk.tsv that is used the older rx.py application.  This list serves to maintain a

continuity with the site regardless of control channel in use.   This same functionality is done within the JSON file used by the newer multi_rx.py application.

boatbod, 2024-06-03:
For control channels that are part of the same system, the main reason to specify anything beyond just the primary control channel in trunk.tsv is if you are monitoring a non-Motorola system that rotates control channel on a regular basis.  Motorola systems always 'prefer' the default cc and as such, unless that isn't available, it's the only one that gets used.

closed 2024-06-06 by boatbod