boatbod/op25 #220: SRC_ADDR not being set in Multi

davcross, 2024-08-12:
When looking at the log output for OP25 using Level 5 or above the SRC_ADDR is with always 0 or NONE.   It should be the radio ID of the unit that was keyed up as it does in the single receiver module.

**To Reproduce**
Start the multi program with a P25 configuration and V 5 or hight.  Look at the output.

I am running on Raspberry Pi 4 with 2 RTL_SDR dongles.   The Dongels get powered from an external USB powered Hub.

boatbod, 2024-08-12:
While well intentioned, I suspect this might not be a bug at all as there are many legitimate reasons for src_addr being 0 or none at various times in a call. Please capture and attach some -v 11 log output showing the incoming signaling containing a source address and the log message you've identified as being problematic.

marklt1, 2024-08-12:
I had a similar experience with some patched talkgroups in which the UID was being reported as 0.
I am not sure if this is your situation but you can try it.

In tk_p25.py line 1405, I removed   "self.talkgroups[tgid]['receiver'] is None" as a condition which restored some of my missing UIDs. I suspected that some of the patched talkgroups were causing that condition to fail.

Original:
def update_talkgroup_srcaddr(self, curr_time, tgid, srcaddr, svcopts=None):
        if (tgid is None or tgid <= 0 or srcaddr is None or srcaddr <= 0 or
            tgid not in self.talkgroups or self.talkgroups[tgid]['receiver'] is None):
            return 0
Updated:
def update_talkgroup_srcaddr(self, curr_time, tgid, srcaddr, svcopts=None):
        if (tgid is None or tgid <= 0 or srcaddr is None or srcaddr <= 0 or
            tgid not in self.talkgroups):
            return 0

davcross, 2024-08-18:
I might see the problem.    I am trying to listen to a Simulcast system.   It appears that they do not pass the srcaddr from one site to the other.   So it appears that unless you are listening to the source ( where the unit is first heard) the srcaddr is not shared across sites.   Or OP25 is simply ignoring it.    I will dig a little deeper as time goes on.

This does not really make sense to me especially if there is patched calls across sites.

marklt1, 2024-08-18:
Simulcast systems shouldn't make a difference, as it is what I mostly listen to. 
The condition I mentioned above may only happen 1% of the time. 
I do export the srcaddr to a listening agent, so I know the value is there.  
I would suggest changing the logging level to at least 5 and then examine the output. 
I was generally able to find what I was looking for there.

boatbod, 2024-08-19:
davecross,
Marklt1 is 100% correct that simulcast (as in multiple RF towers transmitting the same signal) is not a factor in this problem.  Patching might be.  
Supplying the requested "-v 11" debug logs containing instances of src_addr 0 would definitively allow for myself and other to figure out what's going on in your situation.  Without those, any discussion is just idle speculation.
Lets get some logs posted, otherwise I'm going to close out this problem report.
Thanks, Graham

davcross, 2024-08-19:
[x2.log](https://github.com/user-attachments/files/16660573/x2.log)

Here is a log with v > 10

davcross, 2024-08-19:
Here is an additional log 

[x2.log](https://github.com/user-attachments/files/16660962/x2.log)

davcross, 2024-08-19:
Here is an example of the lines where the source address is missing.

08/19/24 14:02:12.584745 [CCP25] set tgid=2351, srcaddr=None, svcopts=None
08/19/24 14:02:12.717226 [CCP25] set tgid=2351, srcaddr=None, svcopts=None
08/19/24 14:02:12.718104 [CCP25] set tgid=2351, srcaddr=None, svcopts=None
08/19/24 14:02:12.850527 [CCP25] set tgid=2351, srcaddr=None, svcopts=None
08/19/24 14:02:12.851770 [CCP25] set tgid=2351, srcaddr=None, svcopts=None
08/19/24 14:02:12.987494 [CCP25] set tgid=2351, srcaddr=None, svcopts=None
08/19/24 14:02:13.109663 [CCP25] set tgid=2351, srcaddr=None, svcopts=None
08/19/24 14:02:13.110541 [CCP25] set tgid=2351, srcaddr=None, svcopts=None
08/19/24 14:02:13.244120 [CCP25] set tgid=2351, srcaddr=None, svcopts=None
08/19/24 14:02:13.378388 [CCP25] set tgid=2351, srcaddr=None, svcopts=None
08/19/24 14:02:13.379355 [CCP25] set tgid=2351, srcaddr=None, svcopts=None
08/19/24 14:02:13.503815 [CCP25] set tgid=2351, srcaddr=None, svcopts=None
08/19/24 14:02:13.637053 [CCP25] set tgid=2351, srcaddr=None, svcopts=None
08/19/24 14:02:13.638497 [CCP25] set tgid=2351, srcaddr=None, svcopts=None
08/19/24 14:02:13.761219 [CCP25] set tgid=2351, srcaddr=None, svcopts=None
08/19/24 14:02:13.774382 [CCP25] set tgid=2351, srcaddr=None, svcopts=None
08/19/24 14:02:13.897166 [CCP25] set tgid=2351, srcaddr=None, svcopts=None
08/19/24 14:02:13.897728 [CCP25] set tgid=2351, srcaddr=None, svcopts=None
08/19/24 14:02:14.029956 [CCP25] set tgid=2351, srcaddr=None, svcopts=None
08/19/24 14:02:14.152955 [CCP25] set tgid=2351, srcaddr=None, svcopts=None

boatbod, 2024-08-19:
When a "grp_v_ch_grant" message (call setup) is received it appears to contain source address (sa) and this is being passed on.
[code]
08/19/24 09:02:12.102252 [0] tsbk(0x00) grp_v_ch_grant: opts: 0x04 freq: 851.062500 ga: 1641 sa: 61500
08/19/24 09:02:12.102887 [CCP25] set tgid=1641, srcaddr=61500, svcopts=0x4
[/code]

However, the "grp_v_ch_grant_up" messages (call in progress update) are of a type that include both ch1 and ch2 frequencies which means there is not enough space to also include a source address, so srcaddr=0.
[code]
08/19/24 09:02:12.103688 [1] tsbk(0x02) grp_v_ch_grant_up: ch1: 860.487500 ga1: 1350 ch2: 860.487500 ga2: 1457
08/19/24 09:02:12.104449 [CCP25] set tgid=1350, srcaddr=None, svcopts=None
[/code]

boatbod, 2024-08-20:
Lastly, if the previous analysis wasn't convincing enough, you can also look in the MAC_PTT messages that are received from the voice channel for source address information.  Some tgids are coming through with them non-zero, but again every single one associated with tgid=1350 has srcaddr=0.
```
gnorbury@yoga2 ~/Downloads/davecross $ grep MAC_PTT: log?-x2.log | grep -v srcaddr=0
log1-x2.log:08/19/24 09:02:27.839651 [0] MAC_PTT: srcaddr=4101018, grpaddr=1641, TDMA slot ID=0, algid=80, keyid=0, mi=00 00 00 00 00 00 00 00 00, rs_errs=1
log1-x2.log:08/19/24 09:02:27.843358 [0] MAC_PTT: srcaddr=4101018, grpaddr=1641, TDMA slot ID=2, algid=80, keyid=0, mi=00 00 00 00 00 00 00 00 00, rs_errs=1
log1-x2.log:08/19/24 09:02:44.223862 [0] MAC_PTT: srcaddr=4101018, grpaddr=1641, TDMA slot ID=6, algid=80, keyid=0, mi=00 00 00 00 00 00 00 00 00, rs_errs=1
log1-x2.log:08/19/24 09:02:44.227454 [0] MAC_PTT: srcaddr=4101018, grpaddr=1641, TDMA slot ID=8, algid=80, keyid=0, mi=00 00 00 00 00 00 00 00 00, rs_errs=1
log1-x2.log:08/19/24 09:03:16.337336 [0] MAC_PTT: srcaddr=3301525, grpaddr=1250, TDMA slot ID=9, algid=80, keyid=0, mi=00 00 00 00 00 00 00 00 00, rs_errs=1
log1-x2.log:08/19/24 09:03:30.753994 [0] MAC_PTT: srcaddr=5401015, grpaddr=1654, TDMA slot ID=9, algid=80, keyid=0, mi=00 00 00 00 00 00 00 00 00, rs_errs=1
log2-x2.log:08/19/24 09:02:27.839651 [0] MAC_PTT: srcaddr=4101018, grpaddr=1641, TDMA slot ID=0, algid=80, keyid=0, mi=00 00 00 00 00 00 00 00 00, rs_errs=1
log2-x2.log:08/19/24 09:02:27.843358 [0] MAC_PTT: srcaddr=4101018, grpaddr=1641, TDMA slot ID=2, algid=80, keyid=0, mi=00 00 00 00 00 00 00 00 00, rs_errs=1
log2-x2.log:08/19/24 09:02:44.223862 [0] MAC_PTT: srcaddr=4101018, grpaddr=1641, TDMA slot ID=6, algid=80, keyid=0, mi=00 00 00 00 00 00 00 00 00, rs_errs=1
log2-x2.log:08/19/24 09:02:44.227454 [0] MAC_PTT: srcaddr=4101018, grpaddr=1641, TDMA slot ID=8, algid=80, keyid=0, mi=00 00 00 00 00 00 00 00 00, rs_errs=1
log2-x2.log:08/19/24 09:03:16.337336 [0] MAC_PTT: srcaddr=3301525, grpaddr=1250, TDMA slot ID=9, algid=80, keyid=0, mi=00 00 00 00 00 00 00 00 00, rs_errs=1
log2-x2.log:08/19/24 09:03:30.753994 [0] MAC_PTT: srcaddr=5401015, grpaddr=1654, TDMA slot ID=9, algid=80, keyid=0, mi=00 00 00 00 00 00 00 00 00, rs_errs=1
log2-x2.log:08/19/24 09:23:05.275539 [0] MAC_PTT: srcaddr=9301556, grpaddr=1457, TDMA slot ID=0, algid=80, keyid=0, mi=00 00 00 00 00 00 00 00 00, rs_errs=1
log2-x2.log:08/19/24 09:32:10.269957 [0] MAC_PTT: srcaddr=101021, grpaddr=1154, TDMA slot ID=8, algid=80, keyid=0, mi=00 00 00 00 00 00 00 00 00, rs_errs=1
```
```
log2-x2.log:08/19/24 09:15:43.032235 [0] MAC_PTT: srcaddr=0, grpaddr=1350, TDMA slot ID=4, algid=80, keyid=0, mi=00 00 00 00 00 00 00 00 00, rs_errs=1
log2-x2.log:08/19/24 09:15:43.035846 [0] MAC_PTT: srcaddr=0, grpaddr=1350, TDMA slot ID=6, algid=80, keyid=0, mi=00 00 00 00 00 00 00 00 00, rs_errs=1
log2-x2.log:08/19/24 09:21:08.094099 [0] MAC_PTT: srcaddr=0, grpaddr=1350, TDMA slot ID=4, algid=80, keyid=0, mi=00 00 00 00 00 00 00 00 00, rs_errs=1
log2-x2.log:08/19/24 09:21:08.097589 [0] MAC_PTT: srcaddr=0, grpaddr=1350, TDMA slot ID=6, algid=80, keyid=0, mi=00 00 00 00 00 00 00 00 00, rs_errs=2
log2-x2.log:08/19/24 09:24:13.562918 [0] MAC_PTT: srcaddr=0, grpaddr=1350, TDMA slot ID=8, algid=80, keyid=0, mi=00 00 00 00 00 00 00 00 00, rs_errs=1
log2-x2.log:08/19/24 09:24:13.694619 [0] MAC_PTT: srcaddr=0, grpaddr=1350, TDMA slot ID=0, algid=80, keyid=0, mi=00 00 00 00 00 00 00 00 00, rs_errs=1
log2-x2.log:08/19/24 09:29:09.518357 [0] MAC_PTT: srcaddr=0, grpaddr=1350, TDMA slot ID=6, algid=80, keyid=0, mi=00 00 00 00 00 00 00 00 00, rs_errs=1
log2-x2.log:08/19/24 09:29:09.523529 [0] MAC_PTT: srcaddr=0, grpaddr=1350, TDMA slot ID=8, algid=80, keyid=0, mi=00 00 00 00 00 00 00 00 00, rs_errs=1
```

boatbod, 2024-08-20:
Closing this issue.  Appears to be RFSS specific, not something wrong with op25.

closed 2024-08-20 by boatbod

davcross, 2024-08-20:
The issue is less than 1/2 of the transmissions have a source.

davcross, 2024-08-20:
Just to understand this.   I am not seeing the MAC_PTT for each transmission.  Is this normal?  

What is the rs_errs=1 at the end of each of these records.

Thanks for you assistance.   It appears that I still cannot use the source address reliably as expected.   

I understand your desire to close these.   But I don't consider this solved.   Does anyone have any other ideas?

davcross, 2024-08-20:
What would the unhandled be 

08/20/24 19:41:54.950047 [0] tdma(0xa8) unhandled: mfid: 0xa4 msg_data: 0xa8a4115044313300000000000000000000
08/20/24 19:41:55.327927 [1] tsbk(0x0f) unhandled: 0x8fa40a80000a000000000000
08/20/24 19:41:55.842309 [1] tsbk(0x0f) unhandled: 0x8fa40a80000a000000000000
08/20/24 19:41:55.995940 [0] tdma(0xa8) unhandled: mfid: 0xa4 msg_data: 0xa8a4115044313300000000000000000000
08/20/24 19:41:56.231397 [1] tsbk(0x38) unhandled: 0xb800080c2c000c2c00010000
08/20/24 19:41:56.368009 [1] tsbk(0x0f) unhandled: 0x8fa40a80000a000000000000
08/20/24 19:41:56.777122 [0] tdma(0xa8) unhandled: mfid: 0xa4 msg_data: 0xa8a4115044313300000000000000000000
08/20/24 19:41:56.890564 [1] tsbk(0x0f) unhandled: 0x8fa40a80000a000000000000
08/20/24 19:41:57.280223 [1] tsbk(0x38) unhandled: 0xb800080c2c000c2c00010000
08/20/24 19:41:57.428222 [1] tsbk(0x0f) unhandled: 0x8fa40a80000a000000000000
08/20/24 19:41:57.825034 [0] tdma(0xa8) unhandled: mfid: 0xa4 msg_data: 0xa8a4115044313300000000000000000000
08/20/24 19:41:57.940045 [1] tsbk(0x0f) unhandled: 0x8fa40a80000a000000000000
08/20/24 19:41:58.467007 [1] tsbk(0x0f) unhandled: 0x8fa40a80000a000000000000
08/20/24 19:41:59.643762 [1] tsbk(0x0f) unhandled: 0x8fa40a80000a000000000000
08/20/24 19:41:59.664262 [0] tdma(0xa8) unhandled: mfid: 0xa4 msg_data: 0xa8a4115044313300000000000000000000
08/20/24 19:42:00.714130 [0] tdma(0xa8) unhandled: mfid: 0xa4 msg_data: 0xa8a4115044313300000000000000000000
08/20/24 19:42:01.746955 [1] tsbk(0x0f) unhandled: 0x8fa40a80000a000000000000
08/20/24 19:42:02.156499 [0] tdma(0xa8) unhandled: mfid: 0xa4 msg_data: 0xa8a4115044313300000000000000000000
08/20/24 19:42:03.444810 [1] tsbk(0x0f) unhandled: 0x8fa40a80000a000000000000
08/20/24 19:42:04.498074 [1] tsbk(0x0f) unhandled: 0x8fa40a80000a000000000000
08/20/24 19:42:05.549380 [1] tsbk(0x38) unhandled: 0xb800080c2c000c2c00010000
08/20/24 19:42:06.331183 [1] tsbk(0x0f) unhandled: 0x8fa40a80000a000000000000
08/20/24 19:42:06.478744 [0] tdma(0xa8) unhandled: mfid: 0xa4 msg_data: 0xa8a4115044313400000000000000000000
08/20/24 19:42:06.592785 [1] tsbk(0x38) unhandled: 0xb800080c2c000c2c00010000
08/20/24 19:42:06.858618 [1] tsbk(0x0f) unhandled: 0x8fa40a80000a000000000000
08/20/24 19:42:07.506504 [1] tsbk(0x0f) unhandled: 0x8fa40a80000a000000000000
08/20/24 19:42:07.776496 [1] tsbk(0x38) unhandled: 0xb800080c2c000c2c00010000
08/20/24 19:42:08.035295 [1] tsbk(0x0f) unhandled: 0x8fa40a80000a000000000000

davcross, 2024-08-21:
I reread my original statement.   There might be some confusion the way I wrote it.  The web page shows the source number.   The debug at level 6 does not show the source number for many conversations.   I am trying to use the log to identify the units that are active at any time.

boatbod, 2024-08-21:
"mfid: 0xa4" means the message is manufacturer proprietary - most of these are not documented in the TIA-102 specs. A limited number of both Harris and Motorola proprietary messages related to patching are understood and decoded, but these unhandled propriety messages are not those. Unhandled simply means the message mfid/opcode combination is not one that op25 understands.   Op25 knows and decodes all the standard 'group voice' messages, but none of the 'user to user',  or data service messages since those play no part in monitoring regular voice traffic. Regardless of whether or not you agree with me, nothing provided in the two -v11 logs shows messages received containing a source address that is not being decided and passed on.  The vast majority of call activity starts with no srcaddr, receives none during the ongoing messaging, and also doesn't have any during teardown.   That more than 50% of calls have no source address is interesting, but only from the point of view that the system you are monitoring is not sending the info over the air in standard P25 messages.  Send me more -v11 logs showing something different and I'll look at them again. On the topic of ph2 MAC_PTTs (similar to ph1 HDUs) the 500-750msec required for a hardware tune followed by locking the sync sequence can cause the initial frame to be missed.  When this happens, the missed data is picked up from later messaging causing an additional delay but not total loss of info.   By far the biggest impact is for encrypted calls due to delays in building ESS data.

davcross, 2024-08-23:
As always I appreciate your time and effort in looking at the logs.

Dave