boatbod/op25 #221: Cannot compile the gr310 branch on Arch/Artix/Manjaro

invader-zimm, 2024-10-12:
**Describe the bug**
On Arch based systems compilation of the gr310 branch fails.

**To Reproduce**
Steps to reproduce the behavior:
1. check out the gr310 branch
2. run `arch-install.sh` or do all necessary steps manually
3. compilation fails with this error:
```
[ 50%] Building CXX object op25/gr-op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/iqfile_source_impl.cc.o
In file included from /usr/include/fmt/format.h:41,
                 from /usr/include/fmt/core.h:5,
                 from /usr/include/spdlog/fmt/fmt.h:28,
                 from /usr/include/gnuradio/io_signature.h:19,
                 from /usr/include/gnuradio/basic_block.h:15,
                 from /usr/include/gnuradio/block.h:17,
                 from /usr/include/gnuradio/sync_block.h:15,
                 from /home/user/Downloads/github/sdr/digital_voice/op25/op25/gr-op25_repeater/lib/../include/gnuradio/op25_repeater/iqfile_source.h:27,
                 from /home/user/Downloads/github/sdr/digital_voice/op25/op25/gr-op25_repeater/lib/iqfile_source_impl.h:26,
                 from /home/user/Downloads/github/sdr/digital_voice/op25/op25/gr-op25_repeater/lib/iqfile_source_impl.cc:28:
/usr/include/fmt/base.h: In instantiation of ‘static void fmt::v11::detail::value<Context>::format_custom_arg(void*, typename Context::parse_context_type&, Context&) [with T = boost::basic_format<char>; Formatter = fmt::v11::formatter<boost::basic_format<char> >; Context = fmt::v11::context; typename Context::parse_context_type = fmt::v11::basic_format_parse_context<char>]’:
/usr/include/fmt/base.h:1383:19:   required from ‘fmt::v11::detail::value<Context>::value(T&) [with T = const boost::basic_format<char>; Context = fmt::v11::context]’
 1383 |     custom.format = format_custom_arg<
      |     ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
 1384 |         value_type, typename Context::template formatter_type<value_type>>;
      |         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/fmt/base.h:1647:41:   required from ‘constexpr fmt::v11::detail::value<Context> fmt::v11::detail::make_arg(T&) [with bool PACKED = true; Context = fmt::v11::context; T = const boost::basic_format<char>; typename std::enable_if<PACKED, int>::type <anonymous> = 0]’
 1647 |   return {arg_mapper<Context>().map(val)};
      |                                         ^
/usr/include/fmt/base.h:2018:74:   required from ‘constexpr fmt::v11::detail::format_arg_store<Context, NUM_ARGS, 0, DESC> fmt::v11::make_format_args(T& ...) [with Context = context; T = {const boost::basic_format<char, std::char_traits<char>, std::allocator<char> >}; long unsigned int NUM_ARGS = 1; long unsigned int NUM_NAMED_ARGS = 0; long long unsigned int DESC = 15; typename std::enable_if<(NUM_NAMED_ARGS == 0), int>::type <anonymous> = 0]’
 2018 |   return {{detail::make_arg<NUM_ARGS <= detail::max_packed_args, Context>(
      |            ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
 2019 |       args)...}};
      |       ~~~~~                                                               
/usr/include/spdlog/logger.h:328:75:   required from ‘void spdlog::logger::log_(spdlog::source_loc, spdlog::level::level_enum, spdlog::string_view_t, Args&& ...) [with Args = {const boost::basic_format<char, std::char_traits<char>, std::allocator<char> >&}; spdlog::string_view_t = fmt::v11::basic_string_view<char>]’
  328 |             fmt::vformat_to(fmt::appender(buf), fmt, fmt::make_format_args(args...));
      |                                                      ~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~
/usr/include/spdlog/logger.h:80:13:   required from ‘void spdlog::logger::log(spdlog::source_loc, spdlog::level::level_enum, fmt::v11::format_string<T ...>, Args&& ...) [with Args = {const boost::basic_format<char, std::char_traits<char>, std::allocator<char> >&}; fmt::v11::format_string<T ...> = fmt::v11::basic_format_string<char, const boost::basic_format<char, std::char_traits<char>, std::allocator<char> >&>]’
   80 |         log_(loc, lvl, details::to_string_view(fmt), std::forward<Args>(args)...);
      |         ~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/spdlog/logger.h:98:12:   required from ‘void spdlog::logger::log(spdlog::source_loc, spdlog::level::level_enum, const T&) [with T = boost::basic_format<char>; typename std::enable_if<(! spdlog::is_convertible_to_any_format_string<const T&>::value), int>::type <anonymous> = 0]’
   98 |         log(loc, lvl, "{}", msg);
      |         ~~~^~~~~~~~~~~~~~~~~~~~~
/usr/include/spdlog/logger.h:90:12:   required from ‘void spdlog::logger::log(spdlog::level::level_enum, const T&) [with T = boost::basic_format<char>]’
   90 |         log(source_loc{}, lvl, msg);
      |         ~~~^~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/spdlog/logger.h:253:12:   required from ‘void spdlog::logger::error(const T&) [with T = boost::basic_format<char>]’
  253 |         log(level::err, msg);
      |         ~~~^~~~~~~~~~~~~~~~~
/home/user/Downloads/github/sdr/digital_voice/op25/op25/gr-op25_repeater/lib/iqfile_source_impl.cc:188:9:   required from here
  188 |         GR_LOG_ERROR(d_logger, boost::format("%s: %s") % filename % strerror(errno));
      |         ^
/usr/include/fmt/base.h:1402:29: error: passing ‘const fmt::v11::formatter<boost::basic_format<char> >’ as ‘this’ argument discards qualifiers [-fpermissive]
 1402 |     ctx.advance_to(cf.format(*static_cast<qualified_type*>(arg), ctx));
      |                    ~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/gnuradio/custom_lock.h:15,
                 from /usr/include/gnuradio/buffer.h:15,
                 from /usr/include/gnuradio/buffer_double_mapped.h:15,
                 from /usr/include/gnuradio/io_signature.h:29:
/usr/include/gnuradio/logger.h:323:10: note:   in call to ‘decltype (((fmt::v11::formatter<boost::basic_format<char> >*)this)->fmt::v11::detail::native_formatter<fmt::v11::basic_string_view<char>, char, fmt::v11::detail::type::string_type>::format<FormatContext>(bfmt.boost::basic_format<char>::str(), ctx)) fmt::v11::formatter<boost::basic_format<char> >::format(const boost::format&, FormatContext&) [with FormatContext = fmt::v11::context; decltype (((fmt::v11::formatter<boost::basic_format<char> >*)this)->fmt::v11::detail::native_formatter<fmt::v11::basic_string_view<char>, char, fmt::v11::detail::type::string_type>::format<FormatContext>(bfmt.boost::basic_format<char>::str(), ctx)) = fmt::v11::basic_appender<char>; boost::format = boost::basic_format<char>]’
  323 |     auto format(const boost::format& bfmt, FormatContext& ctx)
      |          ^~~~~~
make[2]: *** [op25/gr-op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/build.make:244: op25/gr-op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/iqfile_source_impl.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:591: op25/gr-op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
```

**System Information (please complete the following information):**
 - OS: Arch/Artix/Manjaro with all packages up to date
 - GNURadio version: 3.10.11
 - GCC version: 14.2.1 20240910

**Additional context**
Unfortunately I don't know enough about programming to know if the problem comes from op25 or from other sources. Any info on how to fix the issue is greatly appreciated.

oddbondboris, 2024-12-24:
not sure if it works entirely but
https://github.com/robotastic/trunk-recorder/commit/b5e042eb606bbb7cb7d0a0c6e81a0c65ed395116
makes it compile.
not quite sure how that ever worked but i haven't really tested

boatbod, 2024-12-25:
From a support perspective, Arch is a living nightmare and not one I have the time or inclination to deal with.  It's main problem (or "benefit" if you see it that way) is that it is continuously evolving with potentially significant new changes coming at every update.  Fundamentally there is no way to say that "op25 is known to compile and run with Arch Linux ver XX.YY" like we can with Debian/Ubuntu.

Unless you are adept with Linux troubleshooting and coding, I would suggest you switch to one of the more mainstream LTS releases.  e.g  Debian Bookworm, Ubuntu 24.04, Mint 22

closed 2024-12-25 by boatbod