boatbod/op25 #233: when using decryption adp or des if the kid ends in an 0 ex.... dd10,0000,bb10 it comes back with wrong algo using the ./rx.py command

user480511, 2025-03-06:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System Information (please complete the following information):**
 - OS: [e.g. Ununtu 22.04]
 - GNURadio version
 - GCC version

**Additional context**
Add any other context about the problem here.

user480511, 2025-03-07:
sing Python /usr/bin/python3
CPU Features: SSE2+ SSE4.1+ AVX+ FMA+
Using avx for xtrxdsp_iq16_sc32
Using avx for xtrxdsp_iq8_ic16
Using avx for xtrxdsp_iq16_ic16i
Using avx for xtrxdsp_iq8_ic8i
Using avx for xtrxdsp_sc32i_iq16
Using avx for xtrxdsp_iq8_sc32
Using avx for xtrxdsp_iq8_sc32i
Using avx for xtrxdsp_iq16_sc32i
Using avx for xtrxdsp_sc32_iq16
Using avx for xtrxdsp_ic16i_iq16
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.11.0
built-in source types: file fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp xtrx 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
gain: name: LNA range: start 0 stop 49 step 0
setting gain lna to 49
supported sample rates 250000-2560000 step 24000
[R82XX] PLL not locked!
demodulator: xlator if_rate=24000, input_rate=960000, decim=40, if taps=[193,49], resampled_rate=24000, sps=5
op25_audio::open_socket(): enabled udp host(127.0.0.1), wireshark(23456), audio(23456)
p25_frame_assembler_impl: do_imbe[1], do_output[0], do_audio_output[1], do_phase2_tdma[0], do_nocrypt[0]
03/06/25 18:35:04.787182 [0] p25_frame_assembler_impl::control: cmd(set_slotid), args({"tuner": 0, "cmd": "set_slotid", "slotid": 0})
03/06/25 18:35:04.788887 reading crypt_keys file: keys.json
03/06/25 18:35:04.789650 [0] p25_frame_assembler_impl::control: cmd(crypt_key), args({"tuner": 0, "cmd": "crypt_key", "keyid": 0, "algid": 170, "key": [180, 32, 168, 164, 32]})
03/06/25 18:35:04.789724 [0] p25_frame_assembler_impl::control: cmd(crypt_key), args({"tuner": 0, "cmd": "crypt_key", "keyid": 16, "algid": 170, "key": [180, 32, 168, 164, 32]})
03/06/25 18:35:04.789737 [0] op25_crypt_adp::op25_crypt_adp: loading ADP RC4 module
03/06/25 18:35:04.789744 [0] op25_crypt_algs::key: loaded key keyId:0010, algId:aa
03/06/25 18:35:04.789750 [0] op25_crypt_adp::op25_crypt_adp: loading ADP RC4 module
03/06/25 18:35:04.789756 [0] op25_crypt_algs::key: loaded key keyId:0010, algId:aa
03/06/25 18:35:04.789776 [0] p25_frame_assembler_impl::control: cmd(crypt_key), args({"tuner": 0, "cmd": "crypt_key", "keyid": 43536, "algid": 170, "key": [180, 32, 168, 164, 32]})
03/06/25 18:35:04.789808 [0] op25_crypt_algs::key: loaded key keyId:aa10, algId:aa
03/06/25 18:35:04.789834 [0] op25_crypt_algs::key: loaded key keyId:aa10, algId:aa
03/06/25 18:35:04.789863 [0] p25_frame_assembler_impl::control: cmd(crypt_key), args({"tuner": 0, "cmd": "crypt_key", "keyid": 56592, "algid": 129, "key": [128, 128, 128, 128, 128, 8, 128, 128]})
03/06/25 18:35:04.789875 [0] op25_crypt_des::op25_crypt_des: loading DES OFB module
03/06/25 18:35:04.789881 [0] op25_crypt_algs::key: loaded key keyId:dd10, algId:81
03/06/25 18:35:04.789887 [0] op25_crypt_des::op25_crypt_des: loading DES OFB module
03/06/25 18:35:04.789892 [0] op25_crypt_algs::key: loaded key keyId:dd10, algId:81
03/06/25 18:35:04.789912 [0] p25_frame_assembler_impl::control: cmd(crypt_key), args({"tuner": 0, "cmd": "crypt_key", "keyid": 47888, "algid": 129, "key": [128, 128, 128, 128, 128, 128, 136, 136]})
03/06/25 18:35:04.789923 [0] op25_crypt_algs::key: loaded key keyId:bb10, algId:81
03/06/25 18:35:04.789930 [0] op25_crypt_algs::key: loaded key keyId:bb10, algId:81
metadata update not enabled
using ALSA sound system
audio device: default
Listening on 127.0.0.1:23456
python version detected: 3.12.8 (main, Jan 11 2025, 09:42:09) [GCC 14.2.0]
Allocating 15 zero-copy buffers
03/06/25 18:35:07.430881 [0] p25p1_fdma::check_timeout: expired
03/06/25 18:35:08.572364 [0] p25p1_fdma::check_timeout: expired
main: exception occurred
main: exception:
Traceback (most recent call last):

user480511, 2025-03-07:
running the newest version
will try  multi_rx.py haven;t done that yet

boatbod, 2025-03-07:
Here's what's happening, with additional commentary in bold.

{"tuner": 0, "cmd": "crypt_key", "keyid": 0, "algid": 170, "key": [180, 32, 168, 164, 32]}
**keyid=0 is invalid, so nothing is loaded.**

{"tuner": 0, "cmd": "crypt_key", "keyid": 16, "algid": 170, "key": [180, 32, 168, 164, 32]}
loaded key keyId:0010, algId:aa
**correct behavior (16 = 0x0010, 170 = 0xaa)**

{"tuner": 0, "cmd": "crypt_key", "keyid": 43536, "algid": 170, "key": [180, 32, 168, 164, 32]}
loaded key keyId:aa10, algId:aa
**correct behavior (43536 = 0xaa10, 170 = 0xaa)**

{"tuner": 0, "cmd": "crypt_key", "keyid": 56592, "algid": 129, "key": [128, 128, 128, 128, 128, 8, 128, 128]}
loaded key keyId:dd10, algId:81
**correct behavior (56592 = 0xdd10, 129 = 0x81)**

{"tuner": 0, "cmd": "crypt_key", "keyid": 47888, "algid": 129, "key": [128, 128, 128, 128, 128, 128, 136, 136]}
loaded key keyId:bb10, algId:81
**correct behavior (47888 = 0xbb10, 129 = 0x81)**

I'm not seeing the issue

boatbod, 2025-03-08:
Let me know if you have any additional information, otherwise I'm planning to close this out.

closed 2025-04-17 by boatbod