boatbod/op25 #235: I hear nothing when I open scanner

jishie307, 2025-04-16:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System Information (please complete the following information):**
 - OS: [e.g. Ununtu 22.04]
 - GNURadio version
 - GCC version

**Additional context**
Add any other context about the problem here.

boatbod, 2025-04-16:
I'm going to need a whole lot more info to put this in context and offer applicable guidance.
What are you opening and how are you expecting to hear audio?

boatbod, 2025-04-30:
No further information provided. Closing Issue.

closed 2025-04-30 by boatbod