boatbod/op25 #237: Discussion of New UI 2025

boatbod, 2025-04-17:
New UI pull request has been merged into "dev" branch and is currently working for P25 only.  This thread is for ongoing developer discussions pertaining to the new code.

boatbod, 2025-04-17:
**Todo:**
Fix the typo "Synbol" in op25/gr-op25_repeater/www/www-static/index.html
Need to add support for showing active "patches" that was in the legacy ui.
Need to make work with tk_smartnet.
Need to make back-end app use the legacy gui when running under rx.py

**Discussion**
Call History window is interesting to watch, but scrolls quickly on a busy system.  Can we add a configuration option to restrict it only to traffic that op25 tunes a voice channel to?  Rough equivalent would be running with log level -v 1 and seeing the "voice update:" lines in stderr.2

Is there a way to make the boxes draggable/reorderable, or perhaps switch some off?  e.g. Adjacent Sites

triptolemus510, 2025-04-17:
Patches is already there. See screenshot below. The div that contains it is display=none when patches=0

Under Settings, Adjacent Sites can be hidden, Call History and the plot images sizes can be resized.  I can add option to show/hide the frequency table. Draggable / renderable seems like a heavy lift for not much value.

I'm can work on a user option to select from where the Call History table is populated.

[image]

triptolemus510, 2025-04-18:
@boatbod - can you bring over srcaddr and srctag in channel_update? If so, we can display the Call History data consistently in both Frequency and Voice selections.

Also, c_tdma in channel_update - can you explain its purpose/function? It doesn't seem to show up consistently.

See Call History Source selection and implantation on the right.

[image]

boatbod, 2025-04-18:
You should find that srcaddr and srctag are already in the response from the receiver object (voice channel) which comes over in the channel_update["rx_rcvr"] portion of the message.  See tk_p25.py methods rx_ctl::get_chan_status() and p25_receiver::get_status().  Did you need them elsewhere too?  Be aware that srcaddr=0 can be a frequent occurrence depending on timing of receiving a voice grant or update.  You really can't rely on it for much.

Similarly, c_tdma (in main.js) comes from the channel_update["rx_rcvr"]["tdma"] parameter, which is the currently active slot in that voice receiver.  It will be either null, 0, or 1.  An active receiver with c_tdma=null means P25 phase 1 traffic (for P25 protocol).

triptolemus510, 2025-04-19:
Yes. I would like to have srcaddr and srctag in frequency_data. This would provide a consistent display of Call History when it is populated from voice grant or frequency data. 

I recall having the same problems with srcaddr=0 on max branch. c_tdma appears to exhibit similar behavior in that it often shows up null when it should have a slot (?). I don't think c_tdma is very important to show in call history, so I've replaced it with the frequency, which is present consistently in both data sets.

I'm going to push an update to my repo. Changes:

- In Settings, add option to show/hide call history
- In Settings, add option to choose source of call history data (voice grant or frequency data)
- Deleted TDMA/FDMA from Call History, replaced with Frequency
- Log to Call History only when source ID is present

Looking for:

Source ID and Source Tag in json trunk_update under **frequency_data** (see screenshot)
Name of the running Python module somewhere so the js can invoke the legacy ui automatically 

[image]

Noted issues: 
It looks like there might be something wrong with exiting a HOLD talkgroup. Need to look further.
Error display in frequency_data

boatbod, 2025-04-19:
I have added "srcaddrs" array and "srctags" array to the frequency_data[freq] message.  
Valid entries are None (null), or a decimal source address (rid), or a text source tag.
```
            tags = []
            srcaddrs = []
            srctags = []
            for tgid in tgids:
                try:
                    tgid_int = int(tgid)
                    tag = self.talkgroups.get(tgid_int, {}).get('tag', None)
                    srcaddr = self.talkgroups[tgid_int]['srcaddr']
                    srctag = self.get_rid_tag(self.talkgroups[tgid_int]['srcaddr'])
                except (ValueError, TypeError) as e:
                    if self.debug >= 10:
                        sys.stderr.write(f"Error converting TGID '{tgid}' to int: {e}\n")
                    tag = None
                    srcaddr = None
                    srctag = None
                tags.append(tag)
                srcaddrs.append(srcaddr)
                srctags.append(srctag)

```
In addition, I added "satag" and "gatag" to the patch_data message so that you can display tags (if they exist) in the patches window.

triptolemus510, 2025-04-19:
Excellent. I will incorporate those changes and push an update tomorrow (hopefully).

boatbod, 2025-04-19:
I thinking we need still need a better methodology for figuring out call history.  There was a medical consult that just took place which lasted 3 minutes (with some back and forth) that generated 28 entries in the UI call history, vs 5 entries in the stderr.2 log.

Are you deriving Call History from "frequency_data" or "channel_update" messages?  I'm assuming the former, because the latter looks like there's only going to be an entry if a call is being actively received.  Potentially you could look at the list of tgids in frequency_data and see which ones also exist in channel_update.  Those that do are actively tuned.

triptolemus510, 2025-04-20:
The source of the call history data is user choice under Settings.  One is frequency_data and the other is channel_update. Frequency Data is obviously more chatty as you pointed out earlier. 

[image]

The decision to append the Call History table is made in appendCallHistory() and isDuplicate(). This function checks the call history table to determine if a call with the same Talkgroup ID (tgid) and Source ID (sourceId), and same sysid has already been logged within the recent past (defined by MAX_HISTORY_SECONDS) within the MAX_HISTORY_ROWS.

```
		function isDuplicate(tgid, sourceId) {
			const recentRows = tableBody.querySelectorAll("tr");
		
			for (let i = 0; i < Math.min(MAX_HISTORY_ROWS, recentRows.length); i++) {
				const cells = recentRows[i].querySelectorAll("td");
				if (cells.length < 6) continue;
		
				const rowTime   = cells[0].textContent.trim();
				const rowSys    = cells[1].textContent.trim();
				const rowTgid   = cells[3].textContent.trim();
				const rowSrcId  = cells[5].textContent.trim(); // <-- updated to index 5
		
				if (rowSys === sysHex && rowTgid === String(tgid) && rowSrcId === String(sourceId)) {
					const rowDate = new Date();
					const [hours, minutes, seconds] = rowTime.split(':').map(Number);
					rowDate.setHours(hours, minutes, seconds, 0);
		
					const rowEpochSec = Math.floor(rowDate.getTime() / 1000);
					const nowSec = Math.floor(epoch / 1000);
					const delta = Math.abs(nowSec - rowEpochSec);
		
					if (delta <= MAX_HISTORY_SECONDS) {
						return true;
					}
				}
			}
		
			return false;
		}
```

I will give it some further thought...

boatbod, 2025-04-20:
Another Call History observation:-  when set to Voice Grant source, the call history only appears to update from the currently displayed channel.  i.e. if I have a system with 3 channels, only calls for the one that is presently showing in the top window get added to the history log.

boatbod, 2025-04-20:
If it makes things easier, I could create a new response message "call_log" populated at the server end and include as part of the response to an update request, similar to channel_update, trunk_update and frequency_data.  I imagine it would look similar to the 'voice update:' stderr.2 log messages, but json encoded

triptolemus510, 2025-04-20:
call_log sounds intriguing and probably exactly what's needed. How difficult would that be for you to implement? And to confirm, it would send over data intended to be appended to the Call History table?

And you're right, it's populating from just what's displayed.

triptolemus510, 2025-04-20:
gaga and sgtag are coming up null. I see there's a space in the related ga and sg fields. Perhaps that's screwing up the lookup?

[image]

boatbod, 2025-04-20:
I tweaked it; should be working now.
`patch_data: {12001: {12003: {'sg': 12001, 'sgtag': 'CarolineFireDisp', 'ga': 12003, 'gatag': 'CarolineFireTac3'}}}`

triptolemus510, 2025-04-20:
Looks good. I'll post up a www-static update shortly.

[image]

boatbod, 2025-04-20:
I'm just about to commit some preliminary call_log implementation.  It'll looks like this:
`{'json_type': 'call_log', 'log': [{'time': 1745182937.1006382, 'sysid': 1186, 'rcvr': 1, 'prio': 3, 'tgid': 12421, 'tgtag': 'Dorchester EMS Ops 1', 'rid': 2399991, 'rtag': ''}]}

json_type is "call_log" and the "log" payload is a list (array) of log entries.
Each log entry is a dictionary of the following parameters:
"time": epoch time (floating point)
"sysid": decimal system id (matches the system id usually seen as a 3 digit hex)
"rcvr": decimal channel number (zero referenced, same as the Channel +/- button on the upper display)
"prio": decimal call priority (lower is higher priority)
"tgid": decimal Talkgroup Id
"tgtag": text tgid tag
"rid": decimal srcaddr (aka radio id)
"rtag": text rid tag

Conceivably you might receive none, 1, or up to 10 entries in the log payload list.  Once items are sent to the terminal by the app, they are deleted from the local store in the app, so you won't ever receive the same entry more than once.

boatbod, 2025-04-20:
call_log is pushed.  Tested on tk_p25.py.  Probably should work on tk_smartnet.py unless I made a typo, but I suspect some other things don't work there yet.

triptolemus510, 2025-04-20:
Ok, got this mostly working on P25. 

Question: what is happening when a call_log shows up with no rid. Should that log be skipped?

[image]

boatbod, 2025-04-20:
A call_log entry with no RID just means the radio id wasn't know at the time the call was initially tuned.  Many times the RID will be decoded later and added to the terminal display via a channel_update/trunk_update.  These subsequent updates won't trigger extra log entries unless the subscriber let go of the PTT switch and pressed it again.

triptolemus510, 2025-04-20:
So logs should not be skipped when RID = 0, got it.

Going to push an update in a moment. You'll find some new options under Settings, most notably the selector for the Call History source includes Voice Grants (Python) now. Also an option to show the Source ID in the frequency table.

[image]

[image]

boatbod, 2025-04-21:
On the Settings screen there is some debug info in the bottom right.  What is the significance of the "incomplete" and "errors" counters?
```
busy 0 qfull 0 sendq size 0 requests 68
callbacks: total=238 incomplete=171 error=15 OK=52
```

boatbod, 2025-04-21:
Can you correlate the rcrv id to the channel name, or do you need me to include the name in the call_log data?
I also realize I forgot to include 'slot', which could be None, 0 or 1.  I'll add that this evening.

boatbod, 2025-04-22:
At the system level, prio is attached to tgid.  While it is by no means essential, it is nice to see when a higher priority call is received and preempts a lower priority one already in progress.  Your call, if it makes things inconsistent between the different history sources.

triptolemus510, 2025-04-22:
You okay with something like this: 

[image]

boatbod, 2025-04-22:
That'll workG

image.png (view on web)

triptolemus510, 2025-04-22:
Ok, I will push that update shortly.

Here is the info on the debug stuff. All having to do with the web interface. 

busy 0 qfull 0 sendq size 0 requests 3343
callbacks: total=13388 incomplete=10041 error=0 OK=3347

	var html = "busy " + send_busy;
	html += " qfull " + send_qfull;
	html += " sendq size " + send_queue.length;
	html += " requests " + request_count;
	html += "<br>callbacks:";
	html += " total=" + req_cb_count;
	html += " incomplete=" + nfinal_count;
	html += " error=" + n200_count;
	html += " OK=" + r200_count;
	
	
**busy	send_busy**	
This block prevents new requests from being sent while the current one is still processing.
It’s a simple flow control mechanism to avoid flooding the server or overlapping requests.

**qfill - send_queue length > 5**
The command queue length exceeds SEND_QLIMIT = 5;

**sendq size**
The size of the send queue.

**requests	   request_count**
The number of send_command function calls.

**total (callbacks)	req_cb_count**
The number of calls to http_req_cb (once per second as defined in do_onload() )

**incomplete	nfinal_count**
The number of times where http_req_readyState was != DONE (4)

**error	n200_count**
The number of times HTTP result was != HTTP 200 (OK)

**OK	r200_count**
THe number of times the HTTP result was HTTP 200 (OK)

**readyState can be one of the following values:**
0: UNSENT – request has not been opened yet.
1: OPENED – open() has been called.
2: HEADERS_RECEIVED – send() has been called, and headers and status are available.
3: LOADING – downloading; responseText holds partial data.
4: DONE – the operation is complete.

boatbod, 2025-04-22:
Can you add a button in the Settings screen to invoke the command "dump_buffer".  It's somewhat similar to "capture" but dumps the last 5 minutes of received raw symbols from each channel.

triptolemus510, 2025-04-22:
Incomplete is simply logging the progression of a XMLHttpRequest.readyState. You'll notice that it's incremental in steps of 3. http_req_cb is called after open() , so incomplete increments as status 1, status 2 and status 3 are encountered through the process. There was never any data being dropped. On the contrary, that little routine was preventing incomplete data from being processed (because the http req hadn't completed).

XMLHttpRequest is antiquated. Fetch is a cleaner and modern implementation. Most notably, it handles errors much better. You don't have to check readyState manually. Fetch is either pending, fulfilled, or rejected. Because of this, we can't present a nice error message to the user if the server isn't returning anything.

Debug display is cleaned up a little too. The Fetch Errors: 17 you see in the screenshot are the result of stopping OP25 with the UI running.

Will push an update shortly.

[image]

[image]

triptolemus510, 2025-04-22:
> Can you add a button in the Settings screen to invoke the command "dump_buffer". It's somewhat similar to "capture" but dumps the last 5 minutes of received raw symbols from each channel.

Thanks for the reminder. That was on the TODO list. Will get that done next.

boatbod, 2025-04-23:
dump_buffer mostly works behind the scenes, but you might see a log message with "rx_sync::dump_buffer:" in it.
More importantly, you should find your apps directory gifted some ".bin" files after it runs.  One file per channel.
`ch0-dump.bin  ch1-dump.bin  ch2-dump.bin`

triptolemus510, 2025-04-23:
Push tonight is:

Change XMLHttpRequest.readyState method to fetch(), cleaned up HTML debug info
Add dump_buffer button (please test)
Fix tuning button value assignments from hard coded to update with term_config(), see f_tune_button()
Add feature to save the call history table to a CSV file
Removed some blocks of unused code previously commented out

boatbod, 2025-04-23:
I just pushed a couple small fixes; there was a typo in the dump_buffer command being sent, and vertical misalignment of the button column.

triptolemus510, 2025-04-24:
Thank you for those corrections. I've incorporated them.

Can you help me understand what auto_tracking does? I'm looking to implement a little table that summarizes the content of channel_update. Thinking about what might be useful to show here...

[image]

boatbod, 2025-04-24:
The auto tracking code really should be removed, but the frequency error display can stay.
Way back before I implemented an FLL loop in the demodulator, there were several attempts to automatically adjust the ppm and add/subtract a manual frequency offset in order to minimize the tuning error.  It kinda worked, most of the time, but occasionally would just walk away and totally loose the signal.  Down in multi_rx the code is disabled, but the interfaces persist.  When I get a chance, I'll take a peek and see what still needs to be stripped out.

These days the demodulator has a built-in tracker that works more or less automagically.  Get it within a couple kHz and it'll do the tuning on it's own.

triptolemus510, 2025-04-24:
Ok, sounds good. Auto Tracking is out. I'll remove it from the UI.

If I had just one more recommendation of data to bring over, it would be the channel id / band plan information. Over on osmocom, we brought that over in trunk_update. It's a nice add and a super useful tool for non-affiliate scanning. What do you think?

[image]

[image]

boatbod, 2025-04-25:
I've stripped out the excess freq tracking code from multi_rx but it'll still send frequency error info for display in the UI.

Let me take a look at the channel id/band plan.  Clearly that's going to be format specific to P25 vs Smartnet, but I do see advantage in being able to send either type to show the user.

triptolemus510, 2025-04-25:
You're right, cc_event is correct. Looks like I stupidly handled cc_event messages directly in the dispatch_commands() function in main.js. If the lift isn't too heavy, you can send it over in whatever way works best for you, either in a new message type or the existing trunk_update.

Here's what the Channels table looks like. 

[image]

wgbecks, 2025-04-25:
Trip,

Would it be possible to structure the "Presets" similar to how you have setup the Site Aliases such that they're tied to
System IDs?  The object being to facilitate multiple systems/instances that only the presets associated with a given system
will display in the UI for that instance. 

Thanks,

Bill

boatbod, 2025-04-25:
> Would it be possible to structure the "Presets" similar to how you have setup the Site Aliases such that they're tied to System IDs? The object being to facilitate multiple systems/instances that only the presets associated with a given system will display in the UI for that instance.

From a configuration standpoint, the presets are currently defined globally under 'terminal' section of the cfg.json file, but they could be moved into 'trunking' so that they can be defined per-trunked-system (the data structure is keyed off "sysname").  There is already an api method for the terminal to request the full configuration ("get_full_config") - once you have the result from that, you'd just have to parse it appropriately.

triptolemus510, 2025-04-26:
Some new stuff for tonight's update.

Preset scheme is updated per discussion above... moved to 'trunking', UI fetches the config, finds the system from c_name in channel_update, then updates the preset buttons. Preset buttons are hidden when presets are not present. Good suggestion, @wgbecks 

[image]

Added "View Config" button which nicely displays the config file. Find it under Settings.

[image]

The new Channels table is working well, and I think it's super helpful.

[image]

Other minor UI tweaks, including making it just a tad darker.

boatbod, 2025-04-26:
Looking good!
Did you still need me to look at the iden/type/freq/offset/step/slots stuff?  Where will that get displayed?

triptolemus510, 2025-04-26:
Thanks. I've got some more polish and housekeeping to push up, probably Sunday.

Yes. I'd love the band plan to come over. In Osmocom, it displayed just above the frequency table, along with a user option to show/hide it.

[image]

boatbod, 2025-04-26:
When there are fewer than 6 presets defined, switching between trunking systems does not hide the additional buttons.
e.g. Trunking System A has presets AA, AB ,AC while Trunking System B has preset BA.
Display channel 0 (sys A), we see AA, AB, AC, default 4, 5, 6
Display channel 2 (sys B), we see BA, AB, AC, default 4, 5, 6 

[image]

[image]

JoeyAbsi, 2025-04-26:
Just a suggestion for siteAliases: sysID should be in hex format in config.js
Also just throwing the idea here but it would be a plus in the TG file to have a fourth column to specify directly a color (HTML format)

boatbod, 2025-04-26:
> Just a suggestion for siteAliases: sysID should be in hex format in config.js Also just throwing the idea here but it would be a plus in the TG file to have a fourth column to specify directly a color (HTML format), that way it's easier to assign colors to TG when using alpha_tags (that don't usually have FD/PD etc.)

The sysId can already be specified in hex, in the same way as the NAC.  The following are copied directly out of my live config file.
```
               "nac": "0x4a6",
                "wacn": "0xbee00",
                "sysid": "0x4a2",

```
Adding a fourth column to trunk-tags.tsv would be possible, but that would force you to specify something for the third column.  If we get too carried away, we might want to consider changing trunk-tags to .json format to make it more flexible.

triptolemus510, 2025-04-26:
In Osmocom, we used the priority column to specify priority and color. For example, 502 was prio 5 and color 02. It was a pain in the ass and sloppy when trying to decide what color to choose when a color may or may not be present coupled with whether or not SmartColors was enabled.

I think @JoeyAbsi was saying that the Site Aliases (currently residing in the not-so-great location config.js) should be in hex, not dec. sysid is presented to the UI in dec, which is why its that way in config.js -- our time may be better spent figuring a more elegant way to handle site aliases. In Osmcom, is was a separate single json server side which was read/write from the web ui. http_server.py was modified a bit to accomplish this, obviously. I'm not sure if allowing http_server to write was a good idea or not...but it worked.

boatbod, 2025-04-26:
I've given considerable thought to how we might make the config modifiable from the UI, and the only way I can come up with is broadly similar to the osmocom approach.  You'd need a parent process to run the core http server, then spawn/kill/respawn the gnuradio stuff as a child process.  Frankly it's kinda ugly, which is why it hasn't happened yet.

Now with reference to config.js, can't we just add the configurable items to config.json?  Global stuff can go in the 'terminal' section, and sites can be an optional part of the trunking config, much like the presets.

triptolemus510, 2025-04-27:
I think that's probably the best approach for config.js - I will work that angle.

boatbod, 2025-04-27:
In tk_p25, the band plan information derived from various forms of IDEN_UP message is stored on a per p25_system basis in a dictionary structure, which I'm directly inserting into the trunk_update message as 'band_plan'.  It should contain entries as follows:
```
band_plan[iden]                         <<<<< This is the decimal iden
band_plan[iden]['offset']
band_plan[iden]['step']
band_plan[iden]['frequency']
band_plan[iden]['tdma']            <<<<< this is the number of slots per carrier
```
Example:
'band_plan': {
    '3': {'offset': 30000000, 'step': 12500, 'frequency': 762006250, 'tdma': 2}, 
    '0': {'offset': -45000000, 'step': 6250, 'frequency': 851006250}, 
    '1': {'offset': 30000000, 'step': 6250, 'frequency': 762006250}, 
    '2': {'offset': -45000000, 'step': 12500, 'frequency': 851012500, 'tdma': 2}, 
    '4': {'offset': -39000000, 'step': 12500, 'frequency': 935012500}
}

In tk_smartnet the band plan is handled completely differently so don't look for a 'band_plan' entry on smartzone.

triptolemus510, 2025-04-27:
Perfect. P25 only on the band plans.

Think I've got the new site alias implementation working. So the structure is SYSID, RFSS, SITE, Alias. (1F1, 1, 4, ALCO Northwest). Does this align with what you were thinking? 

[image]

triptolemus510, 2025-04-27:
Today's update:

- Incorporate Band Plan with user display preference
- Deprecated config.js - Site Aliases, SmartColors moved to json (attached mine for reference), added bits to parse get_full_config and terminal_config to retrieve them.
- Fixed Preset button issue where default buttons remained displayed when no preset existed for them 
- Improvements to Channels table display, indicates selected channel, highlights holds, add SmartColors.
- Square buttons!
- Add and/or improve View Config, UI Settings, and About displays
- Add "accent color" selector (the text color for values, mostly)
- More js housekeeping (still some to do)

Please test and let me know what doesn't work right.

[ebrcs.json](https://github.com/user-attachments/files/19930091/ebrcs.json)

Channel Table and Band Plan displays:

[image]

UI Settings:

[image]

Config Viewer:

[image]

About:

[image]

boatbod, 2025-04-27:
Something is not working after the most recent merge.
[image]

triptolemus510, 2025-04-27:
hmmm... the settings page should be all new. If you're not seeing this, then something is wrong. I will do a fresh push of index, main css and main js right now.

[image]

boatbod, 2025-04-27:
That's working much better!

triptolemus510, 2025-04-27:
Ok, I will need to set up a config with 2 systems and troubleshoot that.

boatbod, 2025-04-27:
Here is my config for that system

[cfg-first.json](https://github.com/user-attachments/files/19930855/cfg-first.json)

JoeyAbsi, 2025-04-27:
I'm getting the same issue but only monitoring 1 P25 system

triptolemus510, 2025-04-27:
Well, I fixed two other problems with the way Band Plan and Adjacent Sites were handled when multiple systems are present. So that's nice.

Still trying to duplicate the error reported above.

boatbod, 2025-04-27:
The error is being caught in main.js on line 1219, but I don't know which variable was undefined since I don't see any explicit use of the "replace" method around that block of code.

boatbod, 2025-04-27:
Does this help?

[image]

[31 more replies]