boatbod/op25 #245: P25 no advertizeing the bandplan

sonic2wb, 2025-04-27:
So heres the story, our local P25 transmitter died and when they installed the new one. the site is not broadcasting the bandplan for the trunking. ( how i found this out was DSD+ would not work either untill i manually put in the band plan) 

So my question is what do i need to do to get op 25 to work. 

Its decodeing the CC just fine when i set the logging level to 11.

attached is a screenshot of the main webpage.

[image]

i see in trunck.tsv that there is an offset field and this is what i have for DSD+

Network: BEE00.1FC

  Bandplan #0:   Base=851.00625   Offset=+45     Spacing=6.25  BW=12.5

and it seems to work fine. 

Thanks in advance

sonic2wb, 2025-04-27:
also this is part of the log

04/26/25 23:29:29.332808 [0] p25p1_fdma::set_nac: 0x1f4
04/26/25 23:29:29.332838 [0] NAC 0x1f4 TSBK: op=09 : 89 90 0c 80 00 00 00 00 00 00 68 c7
04/26/25 23:29:29.333277 frequency_tracking     135     0       88      0      -88
04/26/25 23:29:29.337041 [0] NAC 0x1f4 TSBK: op=16 : 16 00 00 40 ff ff ff ff 00 01 5f cb
04/26/25 23:29:29.337412 [0] NAC 0x1f4 TSBK: op=02 : 02 00 01 91 25 31 01 91 25 31 74 2f
04/26/25 23:29:29.337626 [0] NAC 0x1f4 TSBK: op=39 : b9 00 02 30 05 7d 04 05 7d 04 25 08
04/26/25 23:29:29.471778 [0] NAC 0x1f4 TSBK: op=3c : 3c 00 00 31 fc 02 38 17 f0 70 88 a5
04/26/25 23:29:29.472400 [0] NAC 0x1f4 TSBK: op=0b : 0b 90 00 00 00 00 00 00 05 7d b8 f8
04/26/25 23:29:29.472637 [0] NAC 0x1f4 TSBK: op=00 : 80 90 cb e1 cb e1 cb e1 cb dd 8c f0
04/26/25 23:29:29.589734 [0] NAC 0x1f4 TSBK: op=30 : 30 00 00 04 6a 12 a5 8c 35 fe e6 d8

boatbod, 2025-04-27:
OP25 currently has no mechanism to override or manually set a P25 band plan as it solely derives this information from various IDEN_UP messages received on the control channel.  Are you sure your system is not broadcasting one of any of these?  Conceivably it might be broadcasting, but for whatever reason, the message might may not be decoded correctly.

Please set log level to "-v 11" in the startup script and run for several minutes, then send me the whole stderr.2 file.

Note: the "offset" field in trunk.tsv does not do what you think it does... it's a frequency offset for sdr dc spike avoidance.

sonic2wb, 2025-04-27:
[will.txt](https://github.com/user-attachments/files/19931563/will.txt)

Here is about a 3min file.    if you need more let me know. 

ive been useing your OP25 for 3+ years now with no issue untill  they changed out the transmitter. 
Hope I can get back to a working state. 

TIA

boatbod, 2025-04-27:
The first thing that comes to mind is you appear to be running an old python2 / gnuradio-3.7 system.  When did you last pull an update from the "gr38" branch of the boatbod repo?  It's quite possible if you are not running fairly recent op25 code that the problem may spontaneously resolve with an application upgrade.

If at all possible, I'd also recommend you upgrade the host to a later operating system release.  It's been a long while since I've seen any gnuradio-3.7.  If I recall, that back around Ubuntu 16.04 or thereabouts.

sonic2wb, 2025-04-27:
I did the pull lastnight, and still the same issue. i used the ./rebuild.sh

sonic2wb, 2025-04-27:
yea the dist is pretty old. it was one of those "its working, dont screw with it" type things

boatbod, 2025-04-28:
Ah ok. I wasn't seeing any "unhandled" messages in the log, which seemed a little unusual since just about every p25 system sends opcodes of some sort that op25 doesn't know how to decode.

So the challenge we face is what to do to work around your crippled p25 system.  I can certainly implement something that will allow you to manually input a band plan, but if I do that, you'll have to switch over to multi_rx.py (from rx.py) because the old app just doesn't have a flexible enough configuration system to support the required change.

Please take a look at the cfg.json file which I pushed today; it is an example config for a single dongle P25 system much like you are running on rx.py.  Make a copy of the file (suitably renamed) and edit the "trunking" section to set the appropriate control channel frequency, then give it a whirl.  Once you get the basic system up and running, I'll work on some code to allow you to attach a "band_plan" to the trunking configuration.  [That may not happen tonight, however]

sonic2wb, 2025-04-28:
OK will do , Thanks for the help. I am currently build a new install on a Raspberry pi 4 with the latest OS . and i will get it up to the point it is now useing mulit_rx

boatbod, 2025-04-28:
I have added some code to dev, master and gr310 branches to read a band_plan structure from the trunking section of multi_rx.py's config.json file.  See the updated example cfg.json (just remove the "#" and populate the data structure with your own system-specific data in your active config file.)

sonic2wb, 2025-04-29:
Traceback (most recent call last):
  File "/home/pi/op25/op25/gr-op25_repeater/apps/./multi_rx.py", line 1083, in <module>
    rx = rx_main()
         ^^^^^^^^^
  File "/home/pi/op25/op25/gr-op25_repeater/apps/./multi_rx.py", line 1045, in __init__
    config = json.loads(open(options.config_file, encoding="utf-8-sig").read())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/decoder.py", line 353, in raw_decode
    obj, end = self.scan_once(s, idx)
               ^^^^^^^^^^^^^^^^^^^^^^
json.decoder.JSONDecodeError: Expecting property name enclosed in double quotes: line 68 column 21 (char 2443)

[image]

sonic2wb, 2025-04-29:
Ok so Update.  I got the json working. for some reason my pi4 setup is not working it cant seen to use the rtlsdr

so i went over the "old"system and it runs but still getting the same behaviour. its decodeing the CC channel but no trunking is happening . 

attached is the log from muli-rx with the cfg1.jason with log level 11

[will1.txt](https://github.com/user-attachments/files/19952369/will1.txt)

[cfg1.json](https://github.com/user-attachments/files/19952373/cfg1.json)

boatbod, 2025-04-29:
On the newly installed rpi4, did you reboot after running the install.sh script?  If not, it may not have loaded the driver blacklist for the rtl devices.  Take a look in /etc/modprobe.d/ to make sure blacklist-rtl.conf exists.  Assuming it is there, run a "dmesg" command to see what the kernel is up to, and also "lsusb" to see what usb devices are available.

For the older system that is able to run multi_rx, it'd be a good idea to remove the hashed out entry in the band_plan structure ("#1") and also the trailing comma on the entry above ("0"). I don't think that will make much difference, but at least it'll be cleaner and correct for what you are trying to do.

Next, I'd like to see a screen shot of your terminal screen and a couple of the plots.  If you are using the new http terminal, please make sure "Show adjacent sites" and "Show P25 band plan" are checked on the Settings page.  For the plots, I'd like to see "Spectrum" and "Mixer".  Here's an example from one of my systems:

[image]

sonic2wb, 2025-04-29:
how do i get that new UI??  when i run the mulirx i get the old ui

sonic2wb, 2025-04-29:
also do you have discord? I could offer you access to the pi's if that would help. my username on discord is ShadowTronics

sonic2wb, 2025-04-29:
here is the log from the pi4 ...  
04/29/25 10:18:08.871683 [0] control channel timeout, freq(859.787500)
04/29/25 10:18:08.872058 [0] set control channel=859.787500
04/29/25 10:18:09.919284 [0] set control channel=859.787500
04/29/25 10:18:10.967916 [0] set control channel=859.787500
04/29/25 10:18:12.021497 [0] control channel timeout, freq(859.787500)
04/29/25 10:18:12.021725 [0] set control channel=859.787500
04/29/25 10:18:13.064166 [0] set control channel=859.787500
04/29/25 10:18:14.117221 [0] set control channel=859.787500
04/29/25 10:18:15.159726 [0] control channel timeout, freq(859.787500)
04/29/25 10:18:15.160302 [0] set control channel=859.787500
04/29/25 10:18:16.214954 [0] set control channel=859.787500
04/29/25 10:18:17.267363 [0] set control channel=859.787500
04/29/25 10:18:18.305305 [0] control channel timeout, freq(859.787500)
04/29/25 10:18:18.305636 [0] set control channel=859.787500
04/29/25 10:18:19.355235 [0] set control channel=859.787500
04/29/25 10:18:20.404065 [0] set control channel=859.787500
04/29/25 10:18:21.453388 [0] control channel timeout, freq(859.787500)
04/29/25 10:18:21.453884 [0] set control channel=859.787500
04/29/25 10:18:22.507566 [0] set control channel=859.787500
04/29/25 10:18:23.549845 [0] set control channel=859.787500
04/29/25 10:18:24.602866 [0] control channel timeout, freq(859.787500)
04/29/25 10:18:24.603102 [0] set control channel=859.787500

Heres the screenshot from the webinterface on the pi4

it seems it cant set the freq for some reason. i am installing SDR++ to make sure the sdr is working

wgbecks, 2025-04-29:
> also do you have discord? I could offer you access to the pi's if that would help. my username on discord is ShadowTronics

@boatbod is quite busy.   I am retired and can help you out if you'd like to email me that you can provide access to the Pi
to offer remote assistance.    

My email address is: wllmbecks@gmail.com 

Bill

sonic2wb, 2025-04-29:
so i just checked and the Pi4 sdr is working correctly so i dont know why op25 is failing to tune it

[image]

sonic2wb, 2025-04-29:
> > also do you have discord? I could offer you access to the pi's if that would help. my username on discord is ShadowTronics
> 
> [@boatbod](https://github.com/boatbod) is quite busy. I am retired and can help you out if you'd like to email me that you can provide access to the Pi to offer remote assistance.
> 
> My email address is: [wllmbecks@gmail.com](mailto:wllmbecks@gmail.com)
> 
> Bill

Id rather just wait on @boatbod .  unless you are read in on my issue.

sonic2wb, 2025-04-29:
awesome im emailing bill now. here is the log from the start up.  it sees the rtl device and seems to be able to access it. 

[newlog.txt](https://github.com/user-attachments/files/19964800/newlog.txt)

sonic2wb, 2025-04-29:
i seem to be getting a new error in the log unless i just missed it 
04/29/25 14:19:54.723001 [0] check_expired_hold: hold_tgid(None), hold_until(0.0)
04/29/25 14:19:54.724263 [0] check_expired_hold: hold_tgid(None), hold_until(0.0)
**04/29/25** 14:19:55.762684 [0] rx_sync::sync_timeout: protocol NONE
04/29/25 14:19:55.765275 [0] control channel timeout, freq(859.787500)

sonic2wb, 2025-04-29:
here is the screen shot of the NEw UI.    so it seems it is useing the rtl sdr as the plot is showing the signal.

[image]

boatbod, 2025-04-29:
Tuning looks good. Curious signal envelope; doesn't look like typical LSM simulcast modulation (see my screenshot above) so possibly a stand-alone site running c4fm (fsk4).  What are your thoughts about that Bill?

sonic2wb, 2025-04-29:
> Tuning looks good. Curious signal envelope; doesn't look like typical LSM simulcast modulation (see my screenshot above) so possibly a stand-alone site running c4fm (fsk4). What are your thoughts about that Bill?

yup is c4fm single site.  i have been meaning to ask do i need to change the demod type in the json to c4fm? also this is on a amateur radio antenna for 144/440mhz. i have a dedication 800/900 yaggi  thats pointed at the tower ( almost line of sight) 
thank you guys so much for helping me out.

wgbecks, 2025-04-29:
I am into the OP’s Pi-4 remotely… CC tuned and is indicating C4FM per the data scope.

boatbod, 2025-04-29:
You may find you need to change the demod_type to fsk4, but I wouldn't change that unless you are seeing lots of errors with the decode.

sonic2wb, 2025-04-29:
You guys are awesome, when we get this working I just spoke with my local group that uses the feed and we want to buy you guys a cup of coffee and or  a beer.

boatbod, 2025-04-30:
As I understand it, an alternative site is providing access to the desired tgids.  I would still like to see if I can make op25 work with the problematic site, but for now this issue can be closed.

closed 2025-04-30 by boatbod