boatbod/op25 #253: Audio dropouts with recent op25 installations

srs4511351, 2025-05-15:
Using a Raspberry Pi 5 with a SDRplay RSP1A SDR.

I recently installed op25 on DietPi OS and Raspberry Pi OS and I am noticing short dropouts in the audio.
I get these with or without using liquidsoap.
Overall CPU load typically runs 10% or less including firefox and file browsers.
Setting the CPU governor to performance reduces the CPU load, but does not improve the dropouts.
My command line to start op25 is
`./rx.py --args 'driver=sdrplay,soapy=0' -N 'IFGR:28,RFGR:1' -S 2000000 -x 1 -f 855.2875e6 -o 5e3 -q -1 -d -200 -T michigantrunk.tsv -V -2 -w 2> stderr-stream0.2`

Older installations of op25 on these OSes are much better.
I tried a Raspberry Pi OS installation with op25 installed on 7/18/2024 and dropouts are few and far between.

Is there a recent change that could cause this?
Can I do something to improve the audio?

----Steve

boatbod, 2025-05-15:
Steve,

There's nothing of any consequence that has recently changed in the rx.py version of boatbod op25, so there's no particular reason why you should experience more audio dropouts now than previously.  All the addition work and data needed by the new UI affect only the multi_rx version of the codebase, with only null stubs added to rx.py and trunking.py for compatibility purposes.  Are you able to pin down the point in the commit history where you're seeing the problem start?

From an optimization standpoint I'd suggest the following:
- Configure rx.py to utilize -O 'pulse' (PulseAudio) audio rather than the default ALSA driver.  Pulse has better buffering and in my view delivers a better experience, but it's not  set as the default because there can be linux permissions issues on headless installs.

- Reduce the sample rate from 2Mhz to 1Mhz (-S 1000000) if this is supported by your sdr hardware.  By doing this you will reduce the length of the FIR filter taps and significantly reduce the cpu overhead both for the usb transfers and the digital filtering & channel selection.

Graham

srs4511351, 2025-05-16:
The command line that does not use liquidsoap does use -O 'pulse'. -O 'alsa' results in no audio on Raspberry Pi OS with pipewire.
I tried this command:
`./rx.py --args 'driver=sdrplay,soapy=0' -N 'IFGR:28,RFGR:0' -S 1000000 -x 1 -f 855.2875e6 -o 5e3 -q -1 -d -200 -T michigantrunk.tsv -V -2 -O 'pulse' -U 2> stderr-stream0.2`
No change in the dropouts.

I have other installations that I can boot to see if they are good. I will try them when I have time.

Note that with Trunk Recorder and Rdio Scanner and a sample rate of 10000000 (10 MHz) works smoothly with no dropouts.

boatbod, 2025-05-16:
Sometimes a clean rebuild has helped resolve non-specific dropout issues.  If it were a systemic problem I'm sure other users would be reporting seeing the same problem.

If the problem is reproducible and can be narrowed down to a specific git commit I'm happy to look in to it, but there's nothing I can do when it's just a broad brush "I'm experiencing dropouts with op25 but this other software I use isn't".

srs4511351, 2025-05-16:
I would like to be able to narrow it down to a specific git commit, but I don't know how to do it with this repository.
Normally I revert to a specific tag with git checkout, but I don't see any tags or releases for op25.

This is occurring on two different newly built systems.
One thing that is common to these systems is that they use packages from deb-multimedia.org and they seem to have upgraded some packages to newer versions.
On the Raspberry Pi, liquidsoap fails with a Segmentation Fault. After working with the liquidsoap maintainers, their recommendation is to use packages from deb-multimedia.org and this fixes liquidsoap.
I can try to revert to the RPI packages and try op25 without liquidsoap to see it is fixes it.

wgbecks, 2025-05-16:
Before expending a lot of time trying to isolate the dropouts to particular past commit, I might suggest a few basics first.   Enable FFT, Constellation, and Mixer plots, one at a time while

op25 is decoding the control channel of your system/site and make screen captures to post here regarding this report.  Next, set logging verbose to (-v 11) then allow op25 to run for long

enough to capture data while the talkgroups of interest have been active, and while audio drop outs are being observed.  Stop op25 and post the logfile in addition to the plots.

boatbod, 2025-05-17:
> I would like to be able to narrow it down to a specific git commit, but I don't know how to do it with this repository.
> Normally I revert to a specific tag with git checkout, but I don't see any tags or releases for op25.

You just git checkout the commit tag shown by "git log".
e.g.
```
gnorbury@yoga9i ~/op25/op25/gr-op25_repeater/apps $ git log
commit 64e7ef5e9c3eb6be9c50c563663ec24ac46c0284 (HEAD -> dev, origin/master, origin/dev, origin/HEAD)
Author: Graham Norbury <gnorbury@bondcar.com>
Date:   Thu May 8 20:55:22 2025 -0400

    Increase http_server threads from default (4) to 6

commit 7f9fb9ffd3f0f22f39ecbf4722b8f3333cf55344
Merge: f9ef8e7 0c0bca2
Author: boatbod <31832374+boatbod@users.noreply.github.com>
Date:   Thu May 8 07:33:16 2025 -0400

    Merge pull request #252 from dunaevai135/patch-tx_c4fm_widepulse
    
    Fix tx after Motorola Widepulse support

commit 0c0bca28c8eaf3863d1b9ccbaf7caf2faa6e127b
Author: Alexey <dunaevai13579@gmail.com>
Date:   Thu May 8 04:57:38 2025 +0400

    Fix tx after Motorola Widepulse support

commit f9ef8e75e9d6bd1004b29379f5fae86e2dee5fb9
Author: Graham Norbury <gnorbury@bondcar.com>
Date:   Thu May 1 07:49:45 2025 -0400

    Another update to attempt to report mid-call srcaddr updates to ui

commit 86290b62e79cdbb94ffc7a729078344930291551
Author: Graham Norbury <gnorbury@bondcar.com>
Date:   Wed Apr 30 20:38:11 2025 -0400

    Fix issue with band_plan preload
```
`git checkout 0c0bca28c8eaf3863d1b9ccbaf7caf2faa6e127b
`

srs4511351, 2025-05-20:
I believe that I found the most recent commit that has minimal audio dropouts.

This has audio dropouts.
```
commit ee4869b9dc8cfb3197f32175e428d674c40a61af (HEAD)
Author: Graham Norbury <gnorbury@bondcar.com>
Date:   Sun Mar 16 16:58:21 2025 -0400

    Fix bug causing loss of first TDMA frame after sync is established
```

Audio dropouts rarely heard, very good.
```
commit 7c3334dc44152dde80a6e580a687ea9f52a04dac (HEAD)
Author: Graham Norbury <gnorbury@bondcar.com>
Date:   Sun Mar 16 16:58:01 2025 -0400

    Fix compiler warning
```
All of the builds that I made that were more recent than commit ee4869b9dc8cfb3197f32175e428d674c40a61af had more than a normal amount of dropouts.

I hope this helps!

boatbod, 2025-05-20:
Based on the very minimal code that was changed between these two commits, are you experiencing the dropouts only on Phase 2 traffic?

As an experiment, would you mind manually editing the file ~/op25/op25/gr-op25_repeater/lib/p25p2_tdma.cc and taking out the two "//" from lines 682 and 683, then building and running the new code?

/* returns true if in sync and slot matches current active slot d_slotid */
int p25p2_tdma::handle_packet(uint8_t dibits[], const uint64_t fs) 
```
{
	// descramble and process the frame
	int rc = -1;
	static const int which_slot[] = {0,1,0,1,0,1,0,1,0,1,1,0};
	packets++;
	sync.check_confidence(dibits);
    **//**if (!sync.in_sync())  // First frame will always be "out of sync" so this check is pretty silly;
    **//**  return -1;          // let other error correction methods prevent bad decode
	const uint8_t* burstp = &dibits[10];
	uint8_t xored_burst[BURST_SIZE - 10];
	burst_type = duid.duid_lookup(duid.extract_duid(burstp));

```

srs4511351, 2025-05-20:
Which version would you like for me to try this on? My current installation is the most recent one that works.

I don't see any way to determine if the individual or current talkgroup is P1 or P2.
All sites on  my system are phase 2 capable, but operate as phase 1, with the exception of Oakland county which is phase 2
and is a Motorola ASTRO-25 700/800 MHz Phase II TRS
I don't see much Oakland county traffic on my system.

srs4511351, 2025-05-20:
I uninstalled op25 and installed the current version after editing ~/op25/op25/gr-op25_repeater/lib/p25p2_tdma.cc as requested.

It seems to have corrected the problem. It is much better and I rarely heard a dropout. It is normal for me to hear infrequent dropouts.

Note that there is a fluttering sound in the audio. I think this has always been there.

boatbod, 2025-05-20:
> Which version would you like for me to try this on? My current installation is the most recent one that works.
 
My goal was for you to try the change against the current version in the repo, which sounds like that's what you did....

> I don't see any way to determine if the individual or current talkgroup is P1 or P2. All sites on my system are phase 2 capable, but operate as phase 1, with the exception of Oakland county which is [...]

The code change affects *only* phase 2 traffic, and only during first frame of a transmission or after some other loss of sync condition.

Can you elaborate on when you would hear the audio dropouts?  Based on the code path, they would seem like they have to be immediately after tuning a new freq, or immediately following a change in PTT (i.e. subscriber releases & re-presses the ptt button, or a different subscriber takes over the conversation).

srs4511351, 2025-05-21:
Yes, I did try the change against the current version in the repo.

According to radioreference.com, all sites on my system are phase 2 capable, but operate as phase 1.
Is is a Phase 2 system that matters? It certainly is better after the edits.

I can hear dropouts at any time. They could be throughout the affected transmission.

srs4511351, 2025-05-21:
I booted up the Raspberry Pi OS system that was also having dropouts.
Sometimes these was a dropout about 1 second  into the transmission. There were also dropouts throughout the transmission.

I uninstalled op25 and installed the patched version.
There are still occasional audio dropouts about 1 second  into a transmission, but dropouts throughout the rest of the transmissions are much less often than with the unpatched version.

boatbod, 2025-05-22:
Based on what I see happening in my test environment, my impression is that your system is experiencing ongoing bit errors in the received data stream, but these are being masked by frames being dropped when sync is lost.  Without the frame dropping code enabled, the errors are passing through and causing spurious speech samples which makes the audio sound glitchy.

If you want to see if this is the case, use the unpatched version of the code and run it at log level -v 9.  Run until you'd had a good number of calls played, then look at the "errs" value for the AMBE codewords in the log.  errs=0 is ideal

wgbecks, 2025-05-22:
Shown below are example error free codewords logged from op25 built on the current  ‘Master Branch’ 

of active Phase-I & Phase-II groups. 

 

IMBE codewords logged from active P25 Phase-I traffic.

05/22/25 12:49:33.742208 [0] IMBE 18 18 5f 3f df b8 04 46 2d 50 c3 errs 0

05/22/25 12:49:33.743773 [0] IMBE 18 67 21 61 40 33 00 b3 be 1c d2 errs 0

05/22/25 12:49:33.745859 [0] IMBE 18 0e f2 d2 f6 5d 00 65 d9 0e c3 errs 0

05/22/25 12:49:33.747451 [0] IMBE 18 70 0f 2f 09 e2 01 9e 26 f1 32 errs 0

05/22/25 12:49:33.749071 [0] IMBE 18 0e f2 d2 f6 5d 00 25 db 0e d3 errs 0

05/22/25 12:50:12.661375 [0] IMBE e3 ac dc d8 aa ca 10 2c 4b 4c 77 errs 0

05/22/25 12:50:12.663441 [0] IMBE 66 7c 90 21 b7 5c 00 5f e7 59 37 errs 0

05/22/25 12:50:12.665428 [0] IMBE 8b 20 ae 7a f4 73 ab 21 e8 82 7b errs 0

05/22/25 12:50:12.666859 [0] IMBE db 21 ad ef 94 4a 64 bc b7 30 60 errs 0

05/22/25 12:50:12.668266 [0] IMBE 39 cf 46 d7 ed d8 d9 51 54 07 98 errs 0

05/22/25 12:50:12.669566 [0] IMBE 15 3d 97 bf 69 42 13 cc dc a9 b0 errs 0

05/22/25 12:50:12.671121 [0] IMBE 08 b7 76 72 e3 a6 e7 c1 a9 04 51 errs 0

 

AMBE codewords logged from active P25 Phase-II traffic.

05/22/25 14:42:44.792194 [0] AMBE a3 3c d1 5a 1b b1 80 errs 0 err_rate 0.000000, dt 1.012332

05/22/25 14:42:44.793084 [0] AMBE a4 1c b1 bf 24 a0 00 errs 0 err_rate 0.000000, dt 0.000890

05/22/25 14:42:44.793590 [0] AMBE a7 06 b8 2d ad e2 80 errs 0 err_rate 0.000000, dt 0.000506

05/22/25 14:42:44.794263 [0] AMBE a8 03 ac 2d ed a7 80 errs 0 err_rate 0.000000, dt 0.000673

05/22/25 14:42:44.953001 [0] AMBE a7 38 35 5f 73 d0 00 errs 0 err_rate 0.000000, dt 0.158738

05/22/25 14:42:44.953551 [0] AMBE a0 86 91 24 c9 ba 80 errs 0 err_rate 0.000000, dt 0.000550

05/22/25 14:42:44.953696 [0] AMBE a0 96 c1 82 e1 2b 00 errs 0 err_rate 0.000000, dt 0.000145

05/22/25 14:42:44.953884 [0] AMBE a0 80 89 99 e9 31 00 errs 0 err_rate 0.000000, dt 0.000188

05/22/25 14:42:45.008395 [0] AMBE a0 80 ab 4c a8 2f 00 errs 0 err_rate 0.000000, dt 0.054511

05/22/25 14:42:45.008755 [0] AMBE 90 85 26 05 af 27 00 errs 0 err_rate 0.000000, dt 0.000360

05/22/25 14:42:45.063553 [0] AMBE 91 59 70 77 a7 a3 00 errs 0 err_rate 0.000000, dt 0.054798

05/22/25 14:42:45.064169 [0] AMBE 91 4b f2 e4 27 e9 00 errs 0 err_rate 0.000000, dt 0.000616

boatbod, 2025-05-23:
No errors in those codewords.  Do you see any when the dropouts occur?

srs4511351, 2025-05-23:
What you are saying makes sense. I can receive this system nearly as well with an indoor antenna, but this data is from using my outdoor discone, at about 20 feet, which provides a much stronger signal.

This transmission had a dropout about 1 second into the transmission. I see errors.
Please also look at the short segment at the end of this message. I noticed that normal sounding transmissions often end with those errors.
```
05/22/25 22:16:39.044027 set tgid=12206, srcaddr=0
05/22/25 22:16:39.044115 hold active tg(12302)
05/22/25 22:16:39.323250 clear tgid=1237, freq=852.687500, slot=0
05/22/25 22:16:39.323307 clear tgid=1237, freq=852.687500, slot=1
05/22/25 22:16:39.417546 set tgid=12206, srcaddr=0
05/22/25 22:16:39.417626 hold active tg(12302)
05/22/25 22:16:39.705190 [0] AMBE 52 d4 67 10 c0 5a 00 errs 6 err_rate 0.145314, dt 4.935119
05/22/25 22:16:39.705277 [0] AMBE d6 d5 56 fd e0 20 80 errs 6 err_rate 0.144432, dt 0.000087
05/22/25 22:16:39.811980 [0] AMBE cd 35 d5 1f 34 ee 00 errs 6 err_rate 0.143595, dt 0.106703
05/22/25 22:16:39.812076 [0] AMBE 51 69 a8 7a e0 a3 00 errs 9 err_rate 0.145991, dt 0.000096
05/22/25 22:16:39.871637 set tgid=12206, srcaddr=0
05/22/25 22:16:40.164421 set tgid=12206, srcaddr=0
05/22/25 22:16:40.547962 set tgid=12206, srcaddr=0
05/22/25 22:16:40.901258 set tgid=12206, srcaddr=0
05/22/25 22:16:41.295052 set tgid=12206, srcaddr=0
05/22/25 22:16:41.416479 set tgid=2173, srcaddr=2009823
05/22/25 22:16:41.416749 voice update:  tg(2173), freq(851175000), slot(-), prio(3)
05/22/25 22:16:41.518361 duid15, tg(2173)
05/22/25 22:16:41.747420 set tgid=2173, srcaddr=0
05/22/25 22:16:41.747582 set tgid=12206, srcaddr=0
05/22/25 22:16:41.747624 hold active tg(2173)
05/22/25 22:16:41.747745 voice update:  tg(2173), freq(851175000), slot(-), prio(3)
05/22/25 22:16:42.092321 [0] IMBE 18 e6 23 e3 40 6b 02 80 fe 80 da errs 0
05/22/25 22:16:42.092985 [0] IMBE 18 f4 07 17 3a bc 02 80 4f c8 7b errs 0
05/22/25 22:16:42.093536 [0] IMBE 18 e4 27 67 58 58 00 8a af fd fa errs 0
05/22/25 22:16:42.094084 [0] IMBE 18 ab b8 58 f7 4d 00 24 03 d7 e3 errs 0
05/22/25 22:16:42.094621 [0] IMBE 18 e2 2b 6b 1b 70 01 97 83 5a 2a errs 0
05/22/25 22:16:42.095166 [0] IMBE 4c ff f8 e7 f8 b7 80 7d b2 19 2d errs 0
05/22/25 22:16:42.095682 [0] IMBE 4d 7e b4 32 9f ba e0 6a a7 8b 74 errs 0
05/22/25 22:16:42.096096 [0] IMBE 51 cd b3 b2 a2 f0 fc 2f ea 13 e1 errs 0
05/22/25 22:16:42.096364 [0] IMBE 51 fc 33 d0 a8 3f e0 11 e7 71 00 errs 0
05/22/25 22:16:42.285597 [0] IMBE 52 ab c5 47 4b c2 e2 63 17 aa 3b errs 0
05/22/25 22:16:42.286049 [0] IMBE 57 64 9a 9c 58 ec fb d6 5f eb 78 errs 0
05/22/25 22:16:42.286229 [0] IMBE 57 5d 42 e1 dd 00 fc ed 0a 5c 0f errs 0
05/22/25 22:16:42.286476 [0] IMBE 5b 69 d5 24 3d 7e fd e5 8a fd 30 errs 0
05/22/25 22:16:42.286656 [0] IMBE 5b 78 95 7c 3a 02 fd d9 ed 0e c3 errs 0
05/22/25 22:16:42.286845 [0] IMBE 5a f5 ca 67 c2 a0 fb da ed de ee errs 0
05/22/25 22:16:42.287029 [0] IMBE 5d f3 59 6d 3e ee f8 f1 05 ad 13 errs 0
05/22/25 22:16:42.287311 [0] IMBE 5d f3 d0 0d 4d 57 f0 31 73 d2 20 errs 0
05/22/25 22:16:42.287747 [0] IMBE 5a 7c 52 31 61 95 e1 e2 97 9d 5f errs 0
05/22/25 22:16:42.288119 set tgid=12206, srcaddr=0
05/22/25 22:16:42.288200 set tgid=12206, srcaddr=0
05/22/25 22:16:42.489241 [0] IMBE 5e 63 8b 6e 9a cb f0 fc f8 fd 5e errs 0
05/22/25 22:16:42.489814 [0] IMBE 62 67 b6 f4 eb 00 f0 66 5a be 53 errs 0
05/22/25 22:16:42.490358 [0] IMBE 62 67 30 f8 57 40 fc 7d 6b a9 82 errs 0
05/22/25 22:16:42.490755 [0] IMBE 62 ce 6e 69 8c b0 fc 0a e9 ff 5b errs 0
05/22/25 22:16:42.491225 [0] IMBE 62 e7 ae 19 5e 04 fc 09 c1 f1 f2 errs 0
05/22/25 22:16:42.491677 [0] IMBE 62 cf 36 b8 d5 2b fc e2 46 5a 0d errs 0
05/22/25 22:16:42.492013 [0] IMBE 62 77 38 38 4b 2a fc fa 61 3d 7c errs 0
05/22/25 22:16:42.492346 [0] IMBE 62 73 9b 1a c3 14 f8 f7 41 ab 47 errs 0
05/22/25 22:16:42.492664 [0] IMBE 65 e3 8f f5 23 d1 f8 f5 72 67 22 errs 0
05/22/25 22:16:42.640894 [0] IMBE 65 4d fa a4 27 fe c0 3e 6d 0c 57 errs 0
05/22/25 22:16:42.641480 [0] IMBE 91 2b 86 93 42 b0 00 09 01 49 40 errs 0
05/22/25 22:16:42.642115 [0] IMBE 18 a9 3d fd 8d 06 02 31 aa ce c3 errs 0
05/22/25 22:16:42.642667 [0] IMBE 18 c2 6b eb 90 69 05 de 95 d9 22 errs 0
05/22/25 22:16:42.643213 [0] IMBE 19 0e 73 f2 84 8f 03 a6 96 66 0b errs 0
05/22/25 22:16:42.643746 [0] IMBE 5a 7f ed f0 7b 23 c0 7f a4 1b 6a errs 0
05/22/25 22:16:42.644185 [0] IMBE 5a 73 ca 05 5d 4f f8 7f f3 37 8d errs 0
05/22/25 22:16:42.644491 [0] IMBE 5e 3f 5d de 30 e7 fb d5 87 08 18 errs 0
05/22/25 22:16:42.644659 [0] IMBE 5d 7a d8 34 1e c7 c8 ef ea df df errs 0
05/22/25 22:16:42.834652 [0] IMBE 5c cd 76 9b 31 2d c0 29 46 a1 0e errs 0
05/22/25 22:16:42.835224 [0] IMBE 9c a0 7f 6b 66 03 00 02 c9 17 c1 errs 0
05/22/25 22:16:42.835923 [0] IMBE 18 e4 27 67 53 b1 01 8d 17 2d 9a errs 0
05/22/25 22:16:42.836808 [0] IMBE 66 5b 46 2d e4 9f 80 11 e1 db a3 errs 0
05/22/25 22:16:42.837365 [0] IMBE 66 61 dd 6f 3f 65 80 1f 3e 41 62 errs 0
[... 52 lines cut ...]
05/22/25 22:16:43.738772 [0] IMBE 81 08 6f 3a e8 bc 00 06 5e c4 41 errs 0
05/22/25 22:16:43.740068 set tgid=12206, srcaddr=0
05/22/25 22:16:43.740161 set tgid=12206, srcaddr=0
05/22/25 22:16:43.898823 [0] IMBE 19 51 0d 95 7d 3c 06 07 eb 5e 4a errs 0
05/22/25 22:16:43.899405 [0] IMBE 18 8d f4 f4 de 28 03 4d c0 f9 43 errs 0
05/22/25 22:16:43.899910 [0] IMBE 19 61 2d c5 67 e7 02 14 3a a5 42 errs 0
05/22/25 22:16:43.900416 [0] IMBE 65 e5 fe c0 44 df 80 02 de 14 53 errs 0
[... cut ...]
05/22/25 22:16:36.610098 [0] IMBE d8 00 00 00 00 00 00 00 00 00 00 errs 0
05/22/25 22:16:36.610697 [0] IMBE fc 00 00 00 00 00 00 00 00 00 00 errs 15
05/22/25 22:16:36.610770 [0] IMBE fc 00 00 00 00 00 00 00 00 00 00 errs 15
05/22/25 22:16:36.610848 [0] IMBE fc 00 00 00 00 00 00 00 00 00 00 errs 15
05/22/25 22:16:36.610867 [0] IMBE fc 00 00 00 00 00 00 00 00 00 00 errs 15
05/22/25 22:16:36.610884 [0] IMBE fc 00 00 00 00 00 00 00 00 00 00 errs 15
05/22/25 22:16:36.610943 [0] IMBE fc 00 00 00 00 00 00 00 00 00 00 errs 15
```

srs4511351, 2025-05-23:
Here is a really long transmission, or maybe two? There were a few dropouts throughout.
It seems that the error lines are AMBE, while the others are IMBE. Why it this?
```
05/22/25 22:27:14.933694 clear tgid=2249, freq=851.862500, slot=0
05/22/25 22:27:14.933760 clear tgid=2249, freq=851.862500, slot=1
05/22/25 22:27:14.944401 set tgid=8040, srcaddr=0
05/22/25 22:27:14.944494 set tgid=22079, srcaddr=0
05/22/25 22:27:14.944536 hold active tg(2249)
05/22/25 22:27:14.944729 set tgid=2249, srcaddr=9583583
05/22/25 22:27:14.944801 hold active tg(2249)
05/22/25 22:27:14.944927 voice update:  tg(2249), freq(851862500), slot(-), prio(3)
05/22/25 22:27:15.045870 duid15, tg(2249)
05/22/25 22:27:15.103469 set tgid=22079, srcaddr=0
05/22/25 22:27:15.103662 set tgid=12303, srcaddr=0
05/22/25 22:27:15.103778 hold active tg(2249)
05/22/25 22:27:15.174687 set tgid=22079, srcaddr=0
05/22/25 22:27:15.174826 set tgid=12303, srcaddr=0
05/22/25 22:27:15.174895 hold active tg(2249)
05/22/25 22:27:15.435616 clear tgid=12062, freq=852.612500, slot=0
05/22/25 22:27:15.435678 clear tgid=12062, freq=852.612500, slot=1
05/22/25 22:27:15.497655 set tgid=2249, srcaddr=0
05/22/25 22:27:15.497749 set tgid=8040, srcaddr=0
05/22/25 22:27:15.497803 hold active tg(2249)
05/22/25 22:27:15.497921 voice update:  tg(2249), freq(851862500), slot(-), prio(3)
05/22/25 22:27:15.748999 [0] IMBE 04 0c fd 7b fb 7d f2 7b 3d 9e 44 errs 0
05/22/25 22:27:15.749127 [0] IMBE 04 0c fd 7b fb 7d f2 7b 3d 9e 44 errs 0
05/22/25 22:27:15.749187 [0] IMBE 04 0c fd 7b fb 7d f2 7b 3d 9e 44 errs 0
05/22/25 22:27:15.749240 [0] IMBE 04 0c fd 7b fb 7d f2 7b 3d 9e 44 errs 0
05/22/25 22:27:15.749291 [0] IMBE 04 0c fd 7b fb 7d f2 7b 3d 9e 44 errs 0
05/22/25 22:27:15.749342 [0] IMBE 04 0c fd 7b fb 7d f2 7b 3d 9e 44 errs 0
05/22/25 22:27:15.749394 [0] IMBE 18 3e 92 32 1e 40 00 24 0b 00 52 errs 0
05/22/25 22:27:15.749918 [0] IMBE b0 49 e9 ee 49 13 00 0f 61 ee f1 errs 0
05/22/25 22:27:15.750473 [0] IMBE 18 3a 1b 17 ad 06 07 04 b7 f4 02 errs 0
05/22/25 22:27:15.941914 [0] IMBE 18 53 49 8d ac 36 07 5f 60 97 a3 errs 0
05/22/25 22:27:15.942481 [0] IMBE 18 2a ba 52 ff 88 07 1e 62 56 b2 errs 0
05/22/25 22:27:15.943255 [0] IMBE 18 3a 1b 27 8f e2 07 35 27 d4 db errs 0
05/22/25 22:27:15.944308 [0] IMBE 18 0b f8 dc ee 8e 07 32 77 9d 92 errs 0
05/22/25 22:27:15.945024 [0] IMBE 18 2b b8 5c 30 2d 06 c1 9d 59 73 errs 0
05/22/25 22:27:15.945618 [0] IMBE a0 86 b8 2a 07 8e 00 0e 95 f1 5c errs 0
05/22/25 22:27:15.946202 [0] IMBE ad 9d 48 a6 46 83 00 07 0b cf 65 errs 0
05/22/25 22:27:15.946776 [0] IMBE 19 a2 3b 23 8b 57 01 8b 93 c0 2a errs 0
05/22/25 22:27:15.947303 [0] IMBE 18 e1 a4 6f c9 a6 07 b8 3f 8f 8b errs 0
05/22/25 22:27:16.113334 [0] IMBE 18 e3 a8 50 3c 68 07 55 e0 35 72 errs 0
05/22/25 22:27:16.113905 [0] IMBE 18 ba 1b 13 3b fc 01 8c c6 20 9b errs 0
05/22/25 22:27:16.114423 [0] IMBE 18 2a ba 5a 2f 2c 07 0f e2 c3 ba errs 0
05/22/25 22:27:16.114944 [0] IMBE 18 72 8a 0e 3e 56 07 45 69 b8 03 errs 0
05/22/25 22:27:16.115453 [0] IMBE 18 3b 19 18 a6 6f 06 6d 71 1a e2 errs 0
05/22/25 22:27:16.115963 [0] IMBE 18 3e 13 13 a8 2e 07 9e e3 11 03 errs 0
05/22/25 22:27:16.116459 [0] IMBE 18 63 a8 48 65 97 07 31 b2 57 72 errs 0
05/22/25 22:27:16.116980 [0] IMBE 18 2a ba 78 d1 69 06 95 8f 79 f3 errs 0
05/22/25 22:27:16.117478 [0] IMBE 18 2b 39 49 f7 cd 07 4f 30 90 b2 errs 0
05/22/25 22:27:16.306680 [0] IMBE 18 bd 84 12 71 9d 02 bc 99 84 53 errs 0
05/22/25 22:27:16.307247 [0] IMBE 18 ad a4 44 61 1f 04 bc c6 df f2 errs 0
05/22/25 22:27:16.307930 [0] IMBE 18 8d f4 a4 48 fa 04 bd be 8d db errs 0
05/22/25 22:27:16.308519 [0] IMBE 18 b8 1f 0f b7 f5 01 2b 77 88 ca errs 0
05/22/25 22:27:16.309131 [0] IMBE 18 bb 98 88 2f 86 04 17 2b fc 63 errs 0
05/22/25 22:27:16.309727 [0] IMBE 18 8a fa 4f a8 3e 06 ee 74 61 12 errs 0
05/22/25 22:27:16.310249 [0] IMBE 76 6e f5 0c 55 bf 80 00 f2 ad 35 errs 0
05/22/25 22:27:16.310766 [0] IMBE 66 e3 e3 93 2a d6 f0 02 6f f8 a4 errs 0
05/22/25 22:27:16.311158 [0] IMBE 5a da 85 0e 38 1f ff ff 0b a9 8b errs 0
05/22/25 22:27:16.440957 clear tgid=12303, freq=851.662500, slot=0
05/22/25 22:27:16.441023 clear tgid=12303, freq=851.662500, slot=1
05/22/25 22:27:16.441053 clear tgid=22079, freq=861.912500, slot=0
05/22/25 22:27:16.441064 clear tgid=22079, freq=861.912500, slot=1
05/22/25 22:27:16.491208 [0] IMBE 5b 78 e4 22 39 b2 ff cb 46 d4 a0 errs 0
05/22/25 22:27:16.491501 [0] IMBE 5a 55 ea 33 77 3a ff fc 40 fb 51 errs 0
05/22/25 22:27:16.491658 [0] IMBE 59 d7 ee 3f d6 e9 fc 16 c8 6f b0 errs 0
05/22/25 22:27:16.491972 [0] IMBE 6e 10 42 9e b4 2f e0 08 3b 69 ef errs 0
05/22/25 22:27:16.492448 [0] IMBE 76 2f be 6f 7e 62 f0 08 20 15 aa errs 0
05/22/25 22:27:16.492894 [0] IMBE 6e 6f 0b 63 6d 21 fe 0f 1d 65 b9 errs 0
[... 283 lines cut ...]
05/22/25 22:27:21.524942 [0] IMBE 18 72 8a 0b fd dc 07 b2 0b 6f 83 errs 0
05/22/25 22:27:21.525439 [0] IMBE 18 d0 c6 9e e9 3e 01 d3 32 b6 22 errs 0
05/22/25 22:27:21.525946 [0] IMBE 68 ee bd 5c 47 5e c0 1d 91 0b 17 errs 0
05/22/25 22:27:21.526424 [0] IMBE 80 82 58 66 9a 6f 00 05 49 36 08 errs 0
05/22/25 22:27:21.684774 [0] IMBE 04 0c fd 7b fb 7d f2 7b 3d 9e 44 errs 0
05/22/25 22:27:21.684885 [0] IMBE 04 0c fd 7b fb 7d f2 7b 3d 9e 45 errs 0
05/22/25 22:27:21.684948 [0] IMBE 04 0c fd 7b fb 7d f2 7b 3d 9e 44 errs 0
05/22/25 22:27:21.685012 [0] IMBE 04 0c fd 7b fb 7d f2 7b 3d 9e 45 errs 0
05/22/25 22:27:21.685066 [0] IMBE 04 0c fd 7b fb 7d f2 7b 3d 9e 44 errs 0
05/22/25 22:27:21.685118 [0] IMBE 04 0c fd 7b fb 7d f2 7b 3d 9e 45 errs 0
[... cut ...]
05/22/25 22:27:21.687316 new tgid=32367  prio 3
05/22/25 22:27:21.757935 duid15, tg(2249)
05/22/25 22:27:21.784041 set tgid=22079, srcaddr=0
05/22/25 22:27:21.784374 set tgid=32367, srcaddr=0
05/22/25 22:27:21.784452 hold active tg(2249)
05/22/25 22:27:21.955018 clear tgid=2249, freq=851.862500, slot=0
05/22/25 22:27:21.955101 clear tgid=2249, freq=851.862500, slot=1
05/22/25 22:27:21.955130 clear tgid=2146, freq=854.187500, slot=0
05/22/25 22:27:21.955145 clear tgid=2146, freq=854.187500, slot=1
```

lwvmobile, 2025-05-23:
I see what's going on, there are random 4V and 2V spurts (phase 2 syncs) during your phase 1 sync, that's why you see random AMBE mixed in there when there should be IMBE only for phase 1. That's your audio drop out. I'm willing to make an educated guess that the commit you found may have acted as a 'gate' to prevent random spurious phase 2 sync patterns from occurring, and now you notice it more frequently.

So, I wonder if you have a noisy band, or co-channel interference, or something going on. Is the signal really strong and your plots tight, or is signal an issue? 

> Note that many transmissions end like this:
>
> ```
> 05/22/25 22:16:36.610098 [0] IMBE d8 00 00 00 00 00 00 00 00 00 00 errs 0
> 05/22/25 22:16:36.610697 [0] IMBE fc 00 00 00 00 00 00 00 00 00 00 errs 15 [...]

Regarding this on the tail end of a phase 1 transmission, I've observed this often enough on some systems, instead of transmitting silence on the end, they just drop the voice entirely, either its dead air, or zero fill, so that's fine if its at the end of the voice transmission.

srs4511351, 2025-05-23:
According to radioreference.com, this is a Phase 2 system, but almost all of the traffic is Phase 1.
If this is causing problems, we need to be able to handle it effectively.

I looked at the control channel with gqrx and it looks very strong and clean.
I can receive this system very well with an indoor antenna placed in a first floor window.

Here are some plots from op25 receiving signals.

[image]
[image]
[image]
[image]

boatbod, 2025-05-23:
The signal appears very clean and tight in the plots, so how the heck can a 2V/4V AMBE frame pop up in the middle of what is essentially a IMBE call?  Well, the answer to that is "it depends" on whether this is rx.py or multi_rx.py.  

In the former case (legacy code) the same datastream is passed to both the p25p1 framer and the p25p2 framer in parallel, then each will independently attempt to build a frame by looking for the frame sync flag.  Once a sync flag is identified, the remainder of the frame will be assembled based on size and then decoded based on the previously identified sync flag.  If the frame contents are garbage data e.g. because the the sync flag has been misidentified, the decode will have excessive FEC errors and the vocoder will handle it like a frame repeat.  Of course the frame being repeated will bear no relation to the actual audio stream and it'll sound like a hiccup, or (best case) injection of silence.

For multi_rx.py the situation is a little different because a common frame assembler tracks and searches for known sync flags and knows what type of datastream it is rebuilding at any given time.  There should be no way that a random TDMA frame could be reconstructed from a FDMA datastream because each frame is only sent to the correct decoder for an in-sync stream.  Conceivably I can envisage a scenario where the frame assembler could swap from FDMA to TDMA and back again, but that would require several consecutive frame sync sequences for everything to sort itself out and start streaming data to the protocol layer.

In this instance I'm pretty sure we are talking about rx.py, so my best guess is that the p25p2 frame assembler saw what it thought was a TDMA sequence, grabbed the correct number of subsequent dibits and then processed it as a 2V or 4V voice frame.

I would be interested to see a symbol capture (raw dibits) when this is happening so I can attempt to reproduce it locally.  The other thing I'd like for the OP to do is set up multi_rx.py and see if he sees the same behavior or improved performance.  I'm not planning or making any significant future enhancements to rx.py because it is legacy code, so migrating to multi_rx.py is something to consider anyway.

Thanks
G.

srs4511351, 2025-05-24:
Here's a short symbol capture. I hope it is enough. There were a few things that went on.

I have it working on multi_rx.py. I have a few questions in (parenthesis)
                "whitelist": "", **(Self-explanitory)**
                "tgid_tags_file": "michigan.tsv", **(Apparently the former TGID Tags File)**
                "rid_tags_file": "", **(Is this the former Blacklist?)**

[op25_symbol_capture.zip](https://github.com/user-attachments/files/20421099/op25_symbol_capture.zip)

boatbod, 2025-05-25:
Here's the original rx.py log fragment from srs4511351 capture file, right when a hiccup occurs.
```
05/23/25 23:33:54.187580 [0] NAC 0x799 LDU2: ESS: algid=80, keyid=0, mi=00 00 00 00 00 00 00 00 00, rs_errs=0
05/23/25 23:33:54.188172 [0] IMBE 27 ce 4b eb 94 f4 fd a8 44 9e 5b errs 0
05/23/25 23:33:54.188296 [0] IMBE 2f 5e 21 5d 36 0c ff 8b 86 a4 fa errs 0
05/23/25 23:33:54.188386 [0] IMBE 36 f9 a0 15 f7 fe fe ae 1a 84 4d errs 0
05/23/25 23:33:54.188489 [0] IMBE 41 cf e1 8b f9 da dc dd 01 41 50 errs 0
05/23/25 23:33:54.188780 [0] IMBE 45 41 2d ff d9 03 84 ef 68 a2 5d errs 0
05/23/25 23:33:54.189214 [0] IMBE 96 43 90 aa ec c4 00 03 90 9c 9c errs 0
05/23/25 23:33:54.189775 [0] IMBE 35 f9 b7 f0 dc 8c 93 d4 48 3a 05 errs 0
05/23/25 23:33:54.190151 [0] IMBE 37 69 53 21 11 e8 fe 2e 3d f7 b4 errs 0
05/23/25 23:33:54.190230 [0] IMBE 37 4b 94 d8 97 df ff e6 45 1f 95 errs 0
05/23/25 23:33:54.195203 nac 799 type 10 state 3 len 0
05/23/25 23:33:54.195307 process_qmsg: P25 info: b'{"encrypted": 0, "algid": 128, "keyid": 0}'
05/23/25 23:33:54.253523 [0] 4V_BURST(0) TDMA slot ID=6 ESS: (partial)
05/23/25 23:33:54.253570 [0] AMBE 80 39 16 56 00 ce 80 errs 8 err_rate 0.008512, dt 2.288067
05/23/25 23:33:54.254254 [0] AMBE 2d 62 63 56 ca c2 80 errs 4 err_rate 0.012342, dt 0.000684
05/23/25 23:33:54.254900 [0] AMBE 2c 95 12 0a f8 0c 00 errs 8 err_rate 0.020237, dt 0.000646
05/23/25 23:33:54.255500 [0] AMBE c2 3a 29 60 7b b2 00 errs 6 err_rate 0.025609, dt 0.000600
05/23/25 23:33:54.381547 [0] NAC 0x799 LDU1: LCW: ec=0, pb=0, sf=0, lco=0 : 00 00 04 00 04 d5 8f 2f 40
05/23/25 23:33:54.381609 [0] IMBE 37 5a 90 b9 5e 0e ff d4 bd 99 7c errs 0
05/23/25 23:33:54.381752 [0] IMBE 37 5a 91 99 eb ac fe 9c fc 89 6f errs 0
05/23/25 23:33:54.381857 [0] IMBE 3a e6 f5 c2 c3 2a ff 07 14 a4 c2 errs 0
05/23/25 23:33:54.381957 [0] IMBE 3d cd 99 d5 6d 59 d9 47 cc ab d3 errs 0
05/23/25 23:33:54.382244 [0] IMBE 45 4d a4 19 bf 05 80 74 fa 6a 24 errs 0
05/23/25 23:33:54.382742 [0] IMBE 48 c1 78 1f 9f 83 80 75 54 54 37 errs 0
05/23/25 23:33:54.383218 [0] IMBE 18 f3 98 2c 4b 32 03 b5 f6 19 3a errs 0
05/23/25 23:33:54.383755 [0] IMBE 18 f2 1b 8b 3c 00 02 34 53 cc 7b errs 0
05/23/25 23:33:54.384335 [0] IMBE 9c f2 69 f1 fb 37 00 09 50 be c2 errs 0
```
Here is the same time set of frames being played back through multi_rx.py.  Notice there is no spurious 4V_BURST.
```
05/24/25 19:39:09.543962 [0] NAC 0x799 LDU2: ESS: algid=80, keyid=0, mi=00 00 00 00 00 00 00 00 00, rs_errs=0
05/24/25 19:39:09.543995 [0] IMBE 27 ce 4b eb 94 f4 fd a8 44 9e 5b errs 0
05/24/25 19:39:09.544091 [0] IMBE 2f 5e 21 5d 36 0c ff 8b 86 a4 fa errs 0
05/24/25 19:39:09.544158 [0] IMBE 36 f9 a0 15 f7 fe fe ae 1a 84 4d errs 0
05/24/25 19:39:09.544230 [0] IMBE 41 cf e1 8b f9 da dc dd 01 41 50 errs 0
05/24/25 19:39:09.544434 [0] IMBE 45 41 2d ff d9 03 84 ef 68 a2 5d errs 0
05/24/25 19:39:09.544716 [0] IMBE 96 43 90 aa ec c4 00 03 90 9c 9c errs 0
05/24/25 19:39:09.545090 [0] IMBE 35 f9 b7 f0 dc 8c 93 d4 48 3a 05 errs 0
05/24/25 19:39:09.545303 [0] IMBE 37 69 53 21 11 e8 fe 2e 3d f7 b4 errs 0
05/24/25 19:39:09.545359 [0] IMBE 37 4b 94 d8 97 df ff e6 45 1f 95 errs 0
05/24/25 19:39:09.545985 [0] check_expired_hold: hold_tgid(None), hold_until(1748129948.468656)
05/24/25 19:39:09.546032 [0] process_qmsg: P25 info: b'{"encrypted": 0, "algid": 128, "keyid": 0}'
05/24/25 19:39:09.723959 [0] NAC 0x799 LDU1: LCW: ec=0, pb=0, sf=0, lco=0 : 00 00 04 00 04 d5 8f 2f 40
05/24/25 19:39:09.724045 [0] IMBE 37 5a 90 b9 5e 0e ff d4 bd 99 7c errs 0
05/24/25 19:39:09.724280 [0] IMBE 37 5a 91 99 eb ac fe 9c fc 89 6f errs 0
05/24/25 19:39:09.724404 [0] IMBE 3a e6 f5 c2 c3 2a ff 07 14 a4 c2 errs 0
05/24/25 19:39:09.724460 [0] IMBE 3d cd 99 d5 6d 59 d9 47 cc ab d3 errs 0
05/24/25 19:39:09.724628 [0] IMBE 45 4d a4 19 bf 05 80 74 fa 6a 24 errs 0
05/24/25 19:39:09.725189 [0] IMBE 48 c1 78 1f 9f 83 80 75 54 54 37 errs 0
05/24/25 19:39:09.725620 [0] IMBE 18 f3 98 2c 4b 32 03 b5 f6 19 3a errs 0
05/24/25 19:39:09.726122 [0] IMBE 18 f2 1b 8b 3c 00 02 34 53 cc 7b errs 0
05/24/25 19:39:09.726463 [0] IMBE 9c f2 69 f1 fb 37 00 09 50 be c2 errs 0
```
Lastly, here is the same set of symbols played through rx.py with a slightly modified p25p2_tdma::handle_packet() method that whines to the log when it thinks there is something wrong with the phase 2 sync.
```
05/24/25 19:53:40.101142 [0] NAC 0x799 LDU2: ESS: algid=80, keyid=0, mi=00 00 00 00 00 00 00 00 00, rs_errs=0
05/24/25 19:53:40.101163 [0] IMBE 27 ce 4b eb 94 f4 fd a8 44 9e 5b errs 0
05/24/25 19:53:40.101262 [0] IMBE 2f 5e 21 5d 36 0c ff 8b 86 a4 fa errs 0
05/24/25 19:53:40.101330 [0] IMBE 36 f9 a0 15 f7 fe fe ae 1a 84 4d errs 0
05/24/25 19:53:40.101404 [0] IMBE 41 cf e1 8b f9 da dc dd 01 41 50 errs 0
05/24/25 19:53:40.101607 [0] IMBE 45 41 2d ff d9 03 84 ef 68 a2 5d errs 0
05/24/25 19:53:40.101889 [0] IMBE 96 43 90 aa ec c4 00 03 90 9c 9c errs 0
05/24/25 19:53:40.102259 [0] IMBE 35 f9 b7 f0 dc 8c 93 d4 48 3a 05 errs 0
05/24/25 19:53:40.102388 nac 799 type 10 state 3 len 0
05/24/25 19:53:40.102471 [0] IMBE 37 69 53 21 11 e8 fe 2e 3d f7 b4 errs 0
05/24/25 19:53:40.102528 [0] IMBE 37 4b 94 d8 97 df ff e6 45 1f 95 errs 0
05/24/25 19:53:40.112521 process_qmsg: P25 info: b'{"encrypted": 0, "algid": 128, "keyid": 0}'
05/24/25 19:53:40.141135 [0] p25p2_tdma::handle_packet(): !sync.in_sync()
[... 1 lines cut ...]
05/24/25 19:53:40.181203 [0] 4V_BURST(0) TDMA slot ID=6 ESS: (partial)
[... cut ...]
Clearly the problem is reproducible, and it confirms my theory that it's an artifact of how the legacy rx.py frame assembler does things in parallel and doesn't consider the bigger picture.  That said, I'm not sure there is an easy fix because if I leave the !sync.in_sync() frame discard in place, we then erroneously throw away the first frame regardless of whether it's a good or bad data stream.

I'm going to need to think about this some more.  The easy solution is to simply stop using rx.py and migrate to multi_rx.py :)
That code is more robust, plus you get to use the new web UI as well!

srs4511351, 2025-05-25:
I am glad that the problem has been made clear to you.
I did manage to get a json file to work with multi_rx.py

While I spent little time closely listening to it, I have yet to hear a single dropout!
I need to try the new web UI, thanks. I think I know how to use it if the json file is like my old example.

There are some things that seem strange in the terminal. The active talkgroup does not appear in the list of frequencies. Near the bottom where it says "talkgroup" there is a number to the far right that doesn't seem to be the talkgroup. I don't see the talkgroup number at all.
I'm still used to the old layout.

boatbod, 2025-05-25:
A fix for the original issue has been pushed to all branches.  That you to lwvmobile for noticing the erroneous tdma bursts, and srs4511351 for taking the time to troubleshoot and send the capture file.

boatbod, 2025-05-25:
> There are some things that seem strange in the terminal. The active talkgroup does not appear in the list of frequencies. Near the bottom where it says "talkgroup" there is a number to the far right [...]

The number on the far right at the bottom of the terminal is the source radio id (i.e. the subscriber that is transmitting).  You can map this to a text string in the rid_tags_file.

In the case of multiple channels being defined, you can move between them in the curses terminal by using the left/right arrow keys.  That said, I've always been a curses terminal guy, but since the new UI came along I rarely use anything but that now.

srs4511351, 2025-05-25:
The new GUI is very nice. I like it.
I have long talkgroup descriptions. The longer ones wrap to a new line and makes the page below that jump.
It is merely annoying, but is there a way to change the page width?

How can I send the audio to be played by the web browser so I can use this system anywhere?

boatbod, 2025-05-25:
> The new version works well with rx.py.
> 
> The new GUI with multi_rx.py is very nice. I like it. I have long talkgroup descriptions. The longer ones wrap to a new line and makes the page below that jump. It is merely annoying, but is there a way to change the page width?
> 
> With View Config, the pull-downs don't stay long enough to read.

You might post here and discuss with the UI author.
https://github.com/boatbod/op25/issues/237

> 
> How can I send the audio to be played by the web browser so I can use this system anywhere?

Right now you can't do that because I haven't come up with a convenient way to stream the audio.  You should also remember that op25's web UI lacks any form of security, so I strongly advise NOT to expose it to the general internet without extra measures such as a vpn or reverse proxy unless you don't mind being hacked.

srs4511351, 2025-05-25:
The talkgroup names are truncated in the curses terminal. I am used to that. I can resize the window to see more of it.
I was talking about the GUI, where long names cause the display to jump down. I do like seeing the whole name, though. The GUI does not resize to the browser window when resized. It is still very nice.

I am using a SDRplay RSP1A SDR. PPM correction seems to work normally in other applications, so I don't think it is the SDR.
I just thought you would like to know. I can get it within 200 Hz, so it's OK for me.

I have other applications that stream audio to the browser, so it is not a priority for me, it's just that op25 is my favorite P25 scanner.