boatbod/op25 #254: Install Guide Help Please

Devinw424, 2025-05-26:
Hello, I have been trying to get op25 up and running for over six months and I keep running into issues left and right I was wondering if anyone would be willing do put together a new install guide? 
I am using Oracle VirtualBox with ubuntu-20.04.6-desktop-amd64 image installed I cant get the rtl-sdr V4 to find the NAC of the system I have used ever command I can find I have messed with the Q value in every direction and I cant get the trunk.tsv file to work right and I cant get the talkgroup.tsv file right either 
I found a video of someone showing how to use Gqrx to make sure the control channel is right and I cant get that to work either I have the right rtl-sdr usb selected and I know what the control channel of the system is.
also I see op25 can use aes-256 if you know the key so how do you put that into op25?
any help would be Greatly appreciated

Devinw424, 2025-05-27:
Hello Thank you for your reply
I believe I have the correct version installed I do believe I did download  'git clone github.com/boatbod/op25 -b gr38'
I was trying to use rx.py since I am only trying to monitor one system. 
I do know the control channel of the system I want to listen to
The system is https://www.radioreference.com/db/sid/6643 the CC is 770.53125c
I would definitely agree most of the existing "guides" to installing op25 are old and I think that's part of my problem 
I am not sure why I cant pick up the NAC from the system also I see the NAC on RR 
not really sure what I am doing wrong 
I really appreciate you taking the time to help me.

boatbod, 2025-05-27:
Please do a guick "git pull" followed by "git status" to make sure you have the latest code.
Then, make yourself a copy of "p25_single_rtl_example.json" (apps directory) and edit line 41 to set the correct control channel frequency.
After that, make yourself a local copy of op25.sh (also in the apps directory) and edit it to refer to your locally modified config file that you created in the step above.
Finally, run the startup script and report back what happens.

If the result is anything less than success, please increase logging to "-v 11" (in the startup script), re-run and then past the content of the stderr.2 log here for us to review.

Devinw424, 2025-05-28:
Boatbod
i made a copy of the op25.sh file and named it "system_test.sh" and it has ./multi_rx.py -v 1 -c my system test.json 2> stderr.2
i did not put the e6 on line 41 of file my system test.json
i made a copy of the p25_single_rtl_example.json and saved it as my system test.json
not sure were to find the logging verbosity
i will tried to launch it and it says bash: ./: Is a directory
i even tried to launch it from dir cd op25/op25/gr-op25_repeater/apps ./system_test.sh and i get "bash: ./: Is a directory"
i will have to pick back up on this tomorrow later in the day.
not sure if you wound be open to do a zoom chat or even a phone call ....i really appreciate all your help.

Devinw424, 2025-05-28:
boatbod 
here is a copy of my file 
[my system test.json](https://github.com/user-attachments/files/20472013/my.system.test.json)

[system_test.json](https://github.com/user-attachments/files/20472020/system_test.json) ****Change File Ext Back to "Sh" it would not let me upload a sh file so i renamed it .json to get it to upload***** but it is sh in my file dir

boatbod, 2025-05-28:
I recommend you not use spaces in filenames.  While they *can* be made to work with proper use of the escape character '\' you'll find it's a whole bunch easier just to make them a continuous string.
e.g. "my system test.json" could more easily be named "my_system_test.json"

If you look back to post #3, wgbecks (Bill) has offered to assist you.  He knows op25 intimately and has helped numerous users get up and running.

Devinw424, 2025-05-28:
Boatbod 
Okay thank you I will rename the files later and see what happens. 
I will get ahold of bill to see about helping out
Can you provide me with any more information of how to input the AES 256 keys? Or dose bill know how to do that?

Devinw424, 2025-05-28:
wgbecks

Would be able to help me out? It would be greatly appreciated

boatbod, 2025-05-28:
> Boatbod Okay thank you I will rename the files later and see what happens. I will get ahold of bill to see about helping out Can you provide me with any more information of how to input the AES 256 keys? Or dose bill know how to do that?

AES keys are the least of you worries.  Once you get the basics working, we can assist with that.

Devinw424, 2025-05-28:
Bill, 
I just sent you a email

boatbod, 2025-05-31:
Underlying cause of this issue was the use of RTL-SDR Blog V4 hardware with the older V3 driver.  Once upgraded to the V4 driver following the “Alternative Debian Package Method” that was outlined in the README @ https://github.com/rtlsdrblog/the driver update was tested to work good with both the new V4 as well as with the older V3.

***********************
Alternative Debian Package Installation Method:
***********************

If you have a system reliant on the Debian packages (eg. FlightRadar24, FlightAware, ADSBExchange images) you can update them directly using this method:
 
sudo apt update
sudo apt install libusb-1.0-0-dev git cmake
sudo apt install debhelper

git clone https://github.com/rtlsdrblog/rtl-sdr-blog

cd rtl-sdr-blog
sudo dpkg-buildpackage -b --no-sign
cd ..
sudo dpkg -i librtlsdr0_*.deb
sudo dpkg -i librtlsdr-dev_*.deb
sudo dpkg -i rtl-sdr_*.deb

closed 2025-05-31 by boatbod